istoreos/istoreos #785: 停掉旁路由istore的DNS后手要仍然能上网

jesse26wong, 2023-05-19:
istoreos (5月12最新固件）做旁路由（192.168.10.2），网关指向ikuai(192.168.10.1)，旁路由DNS指向本身，无法路由器地址是192.168.10.4（无线wifi）。旁路由装有adg(53端口）+mosdns（5335端口）+openclash(7874端口)。局域网各设备由爱快负责dhcp派分地址，网关和dns均指旁路由192.168.10.2。istoreos的DNSMASQ端口改为5399(停用对53监控改为adg)，adg端口保留53，重定向为“无”，设好dns的上游服务器填上127.0.0.1：5335转发到mosdns域名分流，Bootstrap DNS 服务器填移动营运派发dns并添加规则。当将adg和mosdns以及均开启状态时（未设置好openclash），mossh能按规则也能分流（测试方法是分别对国内国外域名添加到白灰名单后对其是否能网页访问），adg也能查看到各设备的dns查询和拦截。
但却发现一个大问题是：当我停用adg或mosdns任何其中一个或全部时，电脑端是都不能访问任可网页，这很正常。但不正常的是：手机设备居然能在停掉上面任何其中一个或两个后，手机均能继续正常上网（关掉手机的移动上网）。寻找解决方法。

jjm2473, 2023-05-19:
你说的是手机浏览器吗，有没有DNS over HTTPS，如果是其他app，那可能本身就直接用ip，而不是通过dns解析ip

jesse26wong, 2023-05-19:
用的是华为手机自带浏览器，也用了qq浏览器试过都是一样。系统自带的测速软件，抖音等等等等都可以用。我有在mosdns设了DOH，但问题是我把mosdns关掉，手机还是可以上网呢？是要在手机上面做什么设置吗?难道这个是我个案来的？有人有没有办法解决？

jjm2473, 2023-05-24:
你看看手机上显示的dns地址是什么