istoreos/istoreos #788: DNS配置错误

chunkingzou, 2023-05-19:
1.关于你要提交的问题
DNS配置错误

 2. 详细叙述

(1) 具体问题

手动配置和自动配置都有问题，可以使用QQ，无法打开网页

### (2) 路由器型号和固件版本

A：电犀牛R68S 21.02.3-2023051217

### (3) 详细日志和/或截图

A：
[image]

[image]

chunkingzou, 2023-05-19:
PPPoe拨号也会出现这种错误

chunkingzou, 2023-05-25:
> 解决方法是DNS配置改手动设置成第一级路由器的网关IP或网络营运商的网关IP可以解决的。
依然不行
[image]

chunkingzou, 2023-05-25:
这个是光猫桥接，上面一个软路由，istoreos接在软路由下面跑docker

z104866, 2023-05-31:
我测试下来好像是adbyby_plus的问题...它一启动...就导致这个问题

LAGOD777, 2023-06-19:
同样的问题，开启 hd，pw，就会 dns 错误，全部关闭就正常

h6fYe34KFSiH, 2023-07-08:
> 我测试下来好像是adbyby_plus的问题...它一启动...就导致这个问题

dnsfilter 也会导致这个问题

Alex-Joe-Lee, 2023-07-15:
> 

这个问题如何解决的？我也遇到了相同的问题，掉电重启以后，dns就报错，目前的解决方案是手动修改/etc/resolv.conf，但是每次都修改会很烦

maoxiaomo, 2023-07-28:
同样的问题

ByZain, 2023-07-28:
怀疑是lan口的dns和ss的dns都配置了，导致循环问答或者抢占之类的以至于国内无法上网。目前发现将lan口的DNS配置指向本身局域网地址(我的是192.168.31.x)，ss的dns正常配置(国外8.8.8.8国内阿里223.5.5.5，绕过大陆模式)，好像可以解决这个问题，已经一天过去了，感觉一切正常，等过两天再来追评。

grock304, 2023-09-11:
我也出现这个问题，最后在 iKuai 中的 DNS 设置里，关闭“强制客户端DNS代理”之后解决了。

不清楚原理，纯粹把 DNS 相关设置试了个遍试出来的。

putianhui, 2023-10-14:
怎么解决啊？  我的j4125 安装istoreOS之后，wan拨号内网设备可以上网，但是istoreOS这个设备dns不能解析啊？  我手动改/etc/resolv.conf里面的dns服务器地址，可以上网了，但是过一段时间就又恢复原样了，web里面的接口dns我手动设置了114和253阿里云dns都不行

samisme0818, 2023-10-18:
> 怎么解决啊？ 我的j4125 安装istoreOS之后，wan拨号内网设备可以上网，但是istoreOS这个设备dns不能解析啊？ 我手动改/etc/resolv.conf里面的dns服务器地址，可以上网了，但是过一段时间就又恢复原样了，web里面的接口dns我手动设置了114和253阿里云dns都不行

我的问题是安装了广告拦截插件，其中有dnsmasq组件，会干预系统的dns配置，禁用就好了。也是找了很久的问题。

jjm2473, 2023-10-26:
@putianhui 可能你的运营商分配的dns有问题，那就手动填写dns。/etc/resolv.conf是自动生成的文件，你修改了也会被程序覆盖。

jjm2473, 2023-10-26:
@chunkingzou 你确定上一级路由器有提供DNS服务？也就是192.168.10.1有DNS服务？有没有试过其他公共DNS，例如114.114.114.114，或者223.5.5.5？

closed 2023-10-26 by jjm2473

auflute, 2023-11-28:
在防火墙的自定义规则里加上：
iptables -t nat -A PREROUTING -p udp --dport 53 -j REDIRECT --to-ports 53
iptables -t nat -A PREROUTING -p tcp --dport 53 -j REDIRECT --to-ports 53
[ -n "$(command -v ip6tables)" ] && ip6tables -t nat -A PREROUTING -p udp --dport 53 -j REDIRECT --to-ports 53
[ -n "$(command -v ip6tables)" ] && ip6tables -t nat -A PREROUTING -p tcp --dport 53 -j REDIRECT --to-ports 53

jjm2473, 2023-12-15:
@auflute 你的方案也行可以解决内网dns错误的问题，但是路由器本身的dns问题还是一样。而且后台DHCP/DNS页面本身就提供了dns劫持开关，不需要自己写防火墙规则。

bingoup886, 2024-02-26:
昨晚遇到同样的问题，但试了试发现是科学工具的问题，bypass没问题，ssrp有问题

QuantumLeap-us, 2024-05-31:
j4125 更新了 passwall  自动的dnsmasq 引起的冲突怎么解决啊

wrb233, 2024-09-16:
> 我测试下来好像是adbyby_plus的问题...它一启动...就导致这个问题

谢谢