istoreos/istoreos #789: istore一些软件无法安装，比如qb等

aicenluo, 2023-05-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue
是，没友类似issue
## 2. 详细叙述

### (1) 具体问题

A：通过iStore [v0.1.13-3]无法安装qb

### (2) 路由器型号和固件版本

A：Intel(R) Celeron(R) J4125 CPU @ 2.00GHz，固件版本：iStoreOS 21.02.3 2023051913

### (3) 详细日志和/或截图

A：
[image]

aicenluo, 2023-05-20:
> 反馈bug/问题模板，提建议请删除
> 
> ## 1.关于你要提交的问题
> 为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue； 注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；
> 
> Q：是否用关键词搜索了issue? (使用 "x" 选择)
> 
> * [ ]  没有类似的issue
>   是，没友类似issue
> 
> ## 2. 详细叙述
> ### (1) 具体问题
> A：通过iStore [v0.1.13-3]无法安装qb，我尝试删除相关安装包后重装，问题仍在。部分其他软件可以安装。请大佬解惑，谢谢！
> 
> ### (2) 路由器型号和固件版本
> A：Intel(R) Celeron(R) J4125 CPU @ 2.00GHz，固件版本：iStoreOS 21.02.3 2023051913
> 
> ### (3) 详细日志和/或截图
> A： [image]

jiefffffggg, 2024-10-07:
> 应该你使用了第三方软件源导致的冲突，不用反馈了，也不可能我们解决

还真是，删掉就好了