istoreos/istoreos #803: iStoreOS 21.02.3 22端口默认开启

YuasKD, 2023-05-26:
一个建议,在ssh访问那里将端口选项默认为lan,当前默认为空,这会导致22端口被暴露到公网,对于新手来说,很危险.
[image]

jjm2473, 2023-05-27:
如果你拨号的话，防火墙会自动禁止wan口访问，不只是22端口，是所有端口。

jjm2473, 2023-05-27:
参考 https://github.com/istoreos/istoreos/discussions/662

closed 2023-05-27 by jjm2473

YuasKD, 2023-05-27:
我在机器上做了个测试当不指定端口时,我使用公网能ssh进来,而指定监听lan口使用公网无法ssh
[image]
[image]

jjm2473, 2023-06-01:
你的网络是拨号还是dhcp的？还有你在测试的时候是在内网还是外网，内网测试的时候就算使用外网ip也不算外网流量。