istoreos/istoreos #805: 是否考虑增加对xray的reality的支持

shaojs321, 2023-05-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
确认没有

注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；
确认没有

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
目前xray4月新出的reality协议速度快，解决了vless容易被封的问题，但软件包中目前还是1.75版本，用不了reality，希望增加1.80版本。

### (1) 具体问题

A：软件包中xray最新1.75，reality需要1.80及以上版本才支持，虽然1.80还是预览版，但已经出了一个多月了，很多人使用反馈很好，速度快，暂时还没出现被封的消息

### (2) 路由器型号和固件版本

A：X86 

### (3) 详细日志和/或截图

A：
[image]
[image]

jjm2473, 2023-05-29:
软件包是openwrt官方提供的，不是我们仓库的，你可以使用离线包来安装 https://github.com/AUK9527/Are-u-ok/tree/main/x86

shaojs321, 2023-05-31:
> 软件包是openwrt官方提供的，不是我们仓库的，你可以使用离线包来安装 https://github.com/AUK9527/Are-u-ok/tree/main/x86

已成功安装

closed 2023-06-08 by jjm2473