istoreos/istoreos #810: 应用过滤在手动上传特征库之后不生效，重启后在应用过滤丢失设备

Ardin963, 2023-05-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
试了好几次，发现只要是手动上传特征库到etc/appfilter之后，不删除原来的就无法识别新的库，如果删除原来的并把新的库改成原库到名字之后能识别到库了，但是运行的时候就不生效，设备选择用户里的用户列表丢失，并且不显示设备当前终端列表的任何设备。

重试多次，有时候可以显示选择用户里的用户列表，但是终端列表显示正在收集数据，重启之后，用户列表和终端设备全都丢失不显示
### (1) 具体问题
更新应用过滤特征库之后，应用过滤无法正常运行
A：

### (2) 路由器型号和固件版本
H68K mini，
iStoreOS 21.02.3 2023052618
r2s用最新固件也会有这个问题
A：

### (3) 详细日志和/或截图

A：

jjm2473, 2023-07-13:
看看最新固件还有没有问题，应用过滤已经同步升级到官方最新的版本了

binfoo, 2023-07-24:
这个我也发现了，默认的应该是feather.cfg，上传的是feather_zh.cfg的样子

closed 2023-07-25 by jjm2473

Ardin963, 2023-07-27:
还是有问题

jjm2473, 2023-07-28:
插件都已经更新到原作者最新的版本了，如果还有问题就只能找原作者了