istoreos/istoreos #812: 旁路由向导ui问题

qzydustin, 2023-06-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时包括已关闭的issue（删掉搜索框的的“is:open”条件）；
Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：打开配置旁路由向导，到达截图一的界面

在Safari中下面的选项框没有加载出来
用Edge可以正常加载

### (2) 路由器型号和固件版本

A：x86_64

[Powered by LuCI istoreos-21.02 branch (git-23.088.34636-f8cc111)](https://github.com/openwrt/luci) / [ArgonTheme v2.2.10.8](https://github.com/jerrykuku/luci-theme-argon) / iStoreOS 21.02.3 2023052616

### (3) 详细日志和/或截图

A：

[image]

jjm2473, 2023-07-13:
啥选项框？LAN口配置方式不是没得选吗

qzydustin, 2023-07-20:
> 啥选项框？LAN口配置方式不是没得选吗

不是，是“提供DHCPv4服务（需要关闭主路由DHCP服务）”和“自动获取IPV6”左边的多选框没有在Safari下显示出来

jjm2473, 2023-10-27:
已修复

closed 2023-10-27 by jjm2473