istoreos/istoreos #819: r5c跑不满千兆

jian263994241, 2023-06-06:
两台r5c 一台刷的istoreos  一台刷的骷髅头固件,   同环境下 istoreos 只能跑到 850mbps 左右 ,   骷髅头的 940mbps 满千兆.  
手上还有另一台 r66s 同款CPU, 刷了istoreos 也只能跑 800mbps左右

jjm2473, 2023-06-08:
没有调优经验，你可以试试测试版的固件有没有改善

GreenTeodoro839, 2023-07-01:
istoreos主要是他们自己的产品用，骷颅头的rockchip固件用了patch加上调度很激进

jjm2473, 2023-07-01:
@GreenTeodoro839 倒也不是，目前我们就出过一款硬件。其他固件可能会超频什么的，具体没研究。

jjm2473, 2023-07-07:
更新到最新版本固件，看看能不能跑满了，我用自建的pppoe起码可以跑到1.3G了

jian263994241, 2023-07-09:
最新版已解决

closed 2023-07-13 by jjm2473

diave971, 2024-11-06:
> 更新到最新版本固件，看看能不能跑满了，我用自建的pppoe起码可以跑到1.3G了

貌似最新版本又出现了相关问题，内网测速只有八百多兆22.03.7 2024080210版本
[image]

GreenTeodoro839, 2024-11-07:
> @diave971 你确定这不是通过wifi测试的？内网有线要么～100M，要么～1000M，要么～2.5G，怎么可能500M到800M。楼主提的问题是pppoe外网速度异常，内网再怎么样也不会有cpu调度瓶颈。

你那是协商速度，而不是实际速度。实际速度和机器性能有关