istoreos/istoreos #820: 畅网8505物理机安装istoreos，cpu无法睿频

jjkle, 2023-06-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：物理机安装istoreos，在系统-杂项设置-cpu调节-cpu性能策略，选中性能模式CPU频率:为2501 MHz，而且策略只有自动和性能模式

### (2) 路由器型号和固件版本

A：畅网8505，固件版本iStoreOS 21.02.3 2023052616

### (3) 详细日志和/或截图

A：
[image]

jjm2473, 2023-06-08:
内核可以添加性能策略，不过睿频应该不属于内核的，可能要在bios里面设置

closed 2023-06-09 by jjm2473

jjkle, 2023-06-09:
[image]
bios默认设置，在win下频率正常，8505 coremark正常分数应该15w左右

jjkle, 2023-06-09:
[image]
8505的coremark跑分应该15w左右，

jjkle, 2023-06-09:
[image]
自动模式下频率基本在600MHz左右
[image]
性能模式下只有第一个线程频率跑到基频

jjm2473, 2023-06-09:
试试这个命令能不能跑满所有核心 `openssl speed -elapsed -evp aes-256-gcm -multi +6`

jjm2473, 2023-06-09:
这个cpu调节基本上只是给arm平台用的，不适合x86，x86的控制比较复杂

jjkle, 2023-06-10:
[image]
openssl speed -elapsed -evp aes-256-gcm -multi +6所有核心能跑起来

jjm2473, 2023-06-15:
那就很明显了啊，coremark没跑满CPU，要么就是你调用coremark的参数不对。

binfoo, 2025-03-30:
巨佬，能否像immoralwrt一样在/status/overview里面加上coremark的跑分，让大家可以直观看到