istoreos/istoreos #822: 最新版本，缺少kmod-nft-compat依赖，还请排查下

bcvb2001, 2023-06-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
有类似的问题，但是去年的，现在又出现了

## 2. 详细叙述

### (1) 具体问题

A：最先0526版本，安装PASSWALL时会提示缺少依赖kmod-nft-compa

### (2) 路由器型号和固件版本

A：r68s

### (3) 详细日志和/或截图

A：
[image]

NanjolnoRing, 2023-06-07:
我R2S刷的最新版也是这个问题，还请大佬排查下

jjm2473, 2023-06-08:
如果能安装过去，就忽略掉。或者使用istore那边的离线安装方式，可能会自动解决这些问题

jjm2473, 2023-06-08:
不要使用最新版本，最新版本适配的openwrt也是最新的。安装之前版本就行了。如果不会自己编译，可以用网友编译好的 https://github.com/AUK9527/Are-u-ok/tree/main

jjm2473, 2023-06-08:
https://github.com/istoreos/istoreos/issues/342

closed 2023-06-08 by jjm2473