istoreos/istoreos #823: iStoreOS Docker 外連問題

2ipm1nd, 2023-06-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：大家好，我安裝好 iStoreOS 之後，將之設定成旁路由，並且安裝了商城裡面的 memos，一切使用良好。但是我現在面臨一個問題，我想要將 memos 設置成外網可以連進並且增加 SSL 的設定。我看了網路上的資料，建議是採用 nginx 來設定，但是在 iStoreOS 內要如何安裝使用？並且將 Docker 與實體網路串接？

### (2) 路由器型号和固件版本

A：
固件版本是istoreos-21.02.3
路由器是X86

### (3) 详细日志和/或截图

A：

jjm2473, 2023-06-09:
既然是旁路由模式，那要在主路由上端口转发进来。至于https，可以自己用docker安装nginx，或者配置uhttpd的ssl（istore安装uhttpd管理界面）

2ipm1nd, 2023-06-11:
感謝！我最後在docker 安裝 Nginx Proxy Manager 然後將埠口 80及 443 改掉之後就能用了，只是SSL只能用自簽而已，算是美中不足但自用足夠了

jjm2473, 2023-06-12:
如果使用 “let‘s encrypt” 的话，SSL有专门的管理工具，也可以自动刷新。

我之前使用过 `linuxserver/swag` docker镜像，主体是nginx，并且集成了证书自动申请和刷新，不过没有 Nginx Proxy Manager 的控制面板。

jjm2473, 2023-06-12:
http方式要求必须是要80端口

2ipm1nd, 2023-06-15:
> http 方式要求必须使用 80 端口，至少在申请时要 80 端口

我原本是旁路由模式，聽了您的建議之後決定改成主路由自己撥號，將 uhttp 的監聽埠口改成 480 跟 4443

```bash
# HTTP listen addresses, multiple allowed

 list listen_http        0.0.0.0:480
 list listen_http        [::]:480

# HTTPS listen addresses, multiple allowed

 list listen_https       0.0.0.0:4443
 list listen_https       [::]:4443
```

重新開機，然後 iStoreOS 登入界面就變成 192.168.100.1:480

接著安裝 Nginx Proxy Manager，安裝完之後不知道為什麼，沒辦法從 `192.168.100.1:81`進入設定介面，反而需要從對外 IP 例如`114.222.2.200:81`才能進入設定畫面

然後透過 iStoreOS 安裝 memos 的 docker，啟動後可以在`192.168.100.1:5230`進入，透過 Nginx Proxy Manager 的設定，我的 DDNS 例如`http://memos.dynv6.net:80`可以進入 memos 的網頁

但使用 Nginx Proxy Manager 內建的 SSL 仍然出現**internal error**的錯誤……。是因為 Nginx Proxy Manager 設定連線的網址不能填`192.168.100.1`而需要填 docker 橋接的`172.6.0.1`嗎？真的很奇怪，當然，自簽的還是可以使用。

不用 DNS 認證是因為免費的 dynv6 貌似沒有提供這個功能

jjm2473, 2023-06-15:
既然是用ssl了，那要使用域名访问才能过ssl的验证，否则浏览器肯定有警告。你在软路由上的“主机名”界面，将域名绑定到本地的ip地址（192.168.100.1）就行了，之后使用域名来访问。

2ipm1nd, 2023-06-15:
搞笑了，原來我之前的設定一直無法從外網連進……，最後我在防火牆→轉發埠口的地方，將 wan80 及 443 轉發到 docker 的 Nginx Proxy Manager 內網址，終於可以正常簽發 SSL 了！！

jjm2473, 2023-06-17:
你的运营商是啥啊，居然愿意开放80和443端口

jjm2473, 2023-06-17:
最简单的还是dns签发，没有端口限制，甚至都不需要配置web服务给签发服务用

closed 2023-07-07 by jjm2473