istoreos/istoreos #827: 挂载移动硬盘，上传文件会掉盘的问题

duojiaqi123, 2023-06-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：通过固态硬盘盒外接固态硬盘，在上传文件时会掉盘

### (2) 路由器型号和固件版本

A：r2s
iStoreOS 21.02.3 2023052616

### (3) 详细日志和/或截图

A：

duojiaqi123, 2023-06-10:
后面还有这种

Sat Jun 10 21:56:17 2023 user.info : luci: accepted login on /admin/quickstart for root from 192.168.100.1
Sat Jun 10 21:57:32 2023 user.info : luci: accepted login on /admin/istorex/router/index for root from 192.168.100.242
Sat Jun 10 21:59:26 2023 kern.info kernel: [87678.090585] usb 1-1: USB disconnect, device number 12
Sat Jun 10 21:59:26 2023 kern.info kernel: [87678.091339] sd 0:0:0:0: Device offlined - not ready after error recovery
Sat Jun 10 21:59:26 2023 kern.err kernel: [87678.104330] blk_update_request: I/O error, dev sda, sector 4000781096 op 0x1:(WRITE) flags 0x800 phys_seg 7 prio class 0
Sat Jun 10 21:59:26 2023 kern.err kernel: [87678.105393] Aborting journal on device sda2-8.
Sat Jun 10 21:59:26 2023 kern.err kernel: [87678.105416] blk_update_request: I/O error, dev sda, sector 41800368 op 0x0:(READ) flags 0x80700 phys_seg 2 prio class 0
Sat Jun 10 21:59:26 2023 kern.err kernel: [87678.106810] blk_update_request: I/O error, dev sda, sector 4000780998 op 0x1:(WRITE) flags 0x800 phys_seg 1 prio class 0
Sat Jun 10 21:59:26 2023 kern.err kernel: [87678.107771] Buffer I/O error on dev sda2, logical block 355, lost sync page write
Sat Jun 10 21:59:26 2023 kern.err kernel: [87678.108483] JBD2: Error -5 detected when updating journal superblock for sda2-8.
Sat Jun 10 21:59:26 2023 kern.err kernel: [87678.108517] blk_update_request: I/O error, dev sda, sector 41799872 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Sat Jun 10 21:59:26 2023 kern.err kernel: [87678.110129] blk_update_request: I/O error, dev sda, sector 1092756224 op 0x0:(READ) flags 0x80700 phys_seg 200 prio class 0
Sat Jun 10 21:59:26 2023 kern.err kernel: [87678.111426] blk_update_request: I/O error, dev sda, sector 128514 op 0x1:(WRITE) flags 0x100000 phys_seg 1 prio class 0
Sat Jun 10 21:59:26 2023 kern.err kernel: [87678.112419] Buffer I/O error on dev sda1, logical block 126466, lost async page write
Sat Jun 10 21:59:26 2023 kern.err kernel: [87678.113239] blk_update_request: I/O error, dev sda, sector 1092756424 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
Sat Jun 10 21:59:26 2023 kern.err kernel: [87678.114321] blk_update_request: I/O error, dev sda, sector 1092756425 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
Sat Jun 10 21:59:26 2023 kern.err kernel: [87678.115321] blk_update_request: I/O error, dev sda, sector 1092756426 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
Sat Jun 10 21:59:26 2023 kern.err kernel: [87678.116625] blk_update_request: I/O error, dev sda, sector 1092756427 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
Sat Jun 10 21:59:26 2023 kern.err kernel: [87678.134133] Buffer I/O error on dev sda1, logical block 126466, lost async page write

jjm2473, 2023-06-19:
是不是电流不够

jjm2473, 2023-06-19:
你的移动硬盘有外接电源的吗？电流是否达标？

closed 2023-07-07 by jjm2473

czz875, 2023-08-27:
r5c也存在该问题，主要是固态硬盘易复现，会产生系统盘和挂载盘同时掉盘的现象，机械偶尔也存在该问题，尝试其他固件无此问题。
[image]

jjm2473, 2023-08-28:
看看磁盘管理那边有没有。还有看看内核日志有没有异常报错

czz875, 2023-08-28:
[  179.222852] print_req_error: 1704 callbacks suppressed
[  179.222861] blk_update_request: I/O error, dev sda, sector 614183232 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 0
[  179.222917] blk_update_request: I/O error, dev sda, sector 614184256 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 0
[  179.222938] blk_update_request: I/O error, dev sda, sector 614185280 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 0
[  179.222957] blk_update_request: I/O error, dev sda, sector 614186304 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 0
[  179.222974] blk_update_request: I/O error, dev sda, sector 614187328 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 0
[  179.222992] blk_update_request: I/O error, dev sda, sector 614188352 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 0
[  179.223009] blk_update_request: I/O error, dev sda, sector 614189376 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 0
[  179.223027] blk_update_request: I/O error, dev sda, sector 614190400 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 0
[  179.223043] blk_update_request: I/O error, dev sda, sector 614191424 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 0
[  179.223061] blk_update_request: I/O error, dev sda, sector 614192448 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 0
[  179.231239] ntfs3: 67 callbacks suppressed
[  179.231247] ntfs3: sda1: failed to read volume at offset 0x1290ea000
[  179.231424] buffer_io_error: 14 callbacks suppressed
[  179.231429] Buffer I/O error on dev sda1, logical block 786647, lost async page write
[  179.522751] ntfs3: sda1: failed to read volume at offset 0x1290ea000
[  179.548093] ntfs3: sda1: failed to read volume at offset 0x1290ea000
[  179.568953] ntfs3: sda1: failed to read volume at offset 0x1290ea000
[  179.588472] ntfs3: sda1: failed to read volume at offset 0x1290ea000
[  179.591574] ntfs3: sda1: failed to read volume at offset 0x1290ea000
[  180.569255] ntfs3: sda1: failed to read volume at offset 0x1290ea000
[  180.569365] ntfs3: sda1: failed to read volume at offset 0x1290ea000
[  193.883227] print_req_error: 402 callbacks suppressed
[  193.883234] blk_update_request: I/O error, dev sda, sector 6293496 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[  193.883282] ntfs3: sda1: failed to read volume at offset 0xbffff000
[  193.883308] blk_update_request: I/O error, dev sda, sector 6295232 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[  193.883318] ntfs3: sda1: failed to read volume at offset 0xc00d8000
[  193.883333] blk_update_request: I/O error, dev sda, sector 6295232 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[  193.883342] ntfs3: sda1: failed to read volume at offset 0xc00d8000
[  193.883354] ntfs3: sda1: ntfs_evict_inode r=0 failed, -22.
[  193.883374] blk_update_request: I/O error, dev sda, sector 6293496 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[  193.883383] ntfs3: sda1: failed to read volume at offset 0xbffff000
[  194.968295] blk_update_request: I/O error, dev sda, sector 6293496 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[  194.968334] ntfs3: sda1: failed to read volume at offset 0xbffff000
[  194.968362] blk_update_request: I/O error, dev sda, sector 6295232 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[  194.968372] ntfs3: sda1: failed to read volume at offset 0xc00d8000
[  194.968386] blk_update_request: I/O error, dev sda, sector 6295232 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[  194.968394] ntfs3: sda1: failed to read volume at offset 0xc00d8000
[  194.968406] ntfs3: sda1: ntfs_evict_inode r=0 failed, -22.
[  194.968425] blk_update_request: I/O error, dev sda, sector 6293496 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[  194.968434] ntfs3: sda1: failed to read volume at offset 0xbffff000

czz875, 2023-08-28:
实测中etx4格式更容易发生问题

jjm2473, 2023-08-28:
格式化以后看看还有没有问题

czz875, 2023-08-30:
做几次实验，大概率是卡 IO 导致 CPU 100% 问题，从而导致读不到盘

czz875, 2023-08-31:
测试了几天，发现是卡i/o了
[image]
这算系统bug吗

jjm2473, 2023-08-31:
系统盘都能掉真是厉害。之后不要看首页显示的状态了，直接在终端执行 `lsblk` 就可以看到磁盘信息

jjm2473, 2023-08-31:
万一首页那边只是因为io卡住没加载出来，那就没有任何参考价值了