istoreos/istoreos #831: Safari 浏览器下终端Web显示异常

vic295, 2023-06-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在Safari浏览器下, 首页点击终端后, 见面没有正常显示提示输入账号/密码. 盲输入账号密码后, 也无法正常显示

### (2) 路由器型号和固件版本

A：x86工控机, iStoreOS 21.02.3 2023052616 / LuCI istoreos-21.02 branch git-23.088.34636-f8cc111

### (3) 详细日志和/或截图
[image]
[image]

A：

vic295, 2023-06-13:
有测试过Chrome 是没有问题的

jjm2473, 2023-06-19:
可能是颜色或者字体的问题？用Chrome就好了。这个工具是ttyd，其中的渲染应该使用的是xterm，你要去这两个项目反馈bug

jjm2473, 2023-07-12:
我用safari测试了下，没有这个问题。怀疑是不是你的浏览器改了什么配置或者插件，导致的异常

closed 2023-07-12 by jjm2473