istoreos/istoreos #833: iStoreOS 21.02.3 2023060914 / LuCI istoreos-21.02 branch git-23.088.34636-f8cc111

ncy92, 2023-06-14:
被控J4215 iStoreOS 21.02.3 2023060914 / LuCI istoreos-21.02 branch git-23.088.34636-f8cc111
DNS不定时失效无法访问网站，DNS只能添加一个不能添加备用DNS
[image]
[image]

ncy92, 2023-06-14:
被控J4125

jjm2473, 2023-06-19:
填个靠谱的dns不就行了，或者看看上游的路由器有没有dns服务

ncy92, 2023-06-19:
> 填个靠谱的dns不就行了，或者看看上游的路由器有没有dns服务

软路由是二级路由，添加主路由拨号获取的IPv6 DNS就报错，现在只用IPv4的DNS没有断网问题

ncy92, 2023-06-19:
[image]
没有改任何配置就报错

closed 2023-07-12 by jjm2473