istoreos/istoreos #836: 急！手动升级出现找不到luci.iform模块，进不了luci页面。

8389899, 2023-06-15:
手动运行升级软件后
opkg list-upgradable | awk -F ' - ' '{print $1}' | xargs opkg upgrade
出现进不了luci，提示如下：
/usr/lib/lua/luci/controller/systools.lua:4: module 'luci.iform' not found:
	no field package.preload['luci.iform']
	no file './luci/iform.lua'
	no file '/usr/share/lua/luci/iform.lua'
	no file '/usr/share/lua/luci/iform/init.lua'
	no file '/usr/lib/lua/luci/iform.lua'
	no file '/usr/lib/lua/luci/iform/init.lua'
	no file './luci/iform.so'
	no file '/usr/lib/lua/luci/iform.so'
	no file '/usr/lib/lua/loadall.so'
	no file './luci.so'
	no file '/usr/lib/lua/luci.so'
	no file '/usr/lib/lua/loadall.so'
stack traceback:
	[C]: in function 'require'
	/usr/lib/lua/luci/controller/systools.lua:4: in main chunk
	[C]: in function 'require'
	/usr/lib/lua/luci/dispatcher.lua:1111: in function 'createindex'
	/usr/lib/lua/luci/dispatcher.lua:1211: in function 'createtree'
	/usr/lib/lua/luci/dispatcher.lua:705: in function 'menu_json'
	/usr/lib/lua/luci/dispatcher.lua:806: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:478: in function </usr/lib/lua/luci/dispatcher.lua:477>

软路由机型是X86
进不了luci了，要怎么还原回去？

jjm2473, 2023-06-19:
重刷，谁让你乱升级的。

closed 2023-06-19 by jjm2473

8389899, 2023-06-19:
有没有办法退回升级前的状态？

jjm2473, 2023-06-20:
除非你用了沙箱，那很简单，退出沙箱即可。

还有systool的系统软件修复也可以试试，但是现在你都已经打不开luci了，没法操作，除非自己手动执行。

8389899, 2023-06-20:
麻烦大佬能不能说下怎么手动执行？非常感谢。

jjm2473, 2023-07-17:
很麻烦，而且你之前也不一定安装了systool，还是恢复出厂设置吧，别浪费时间了

8389899, 2023-07-17:
先谢谢大佬了！
systool已经安装了，我记得systool里有重装功能，现在luci进不去没法用了，手动方式要怎么做，能否麻烦大佬解惑？

jjm2473, 2023-07-17:
直接执行 `/usr/share/systools/reset_rom_pkgs.run`，大概10秒后会自动重启

jjm2473, 2023-07-17:
如果不知道哪里执行命令，那就打开后台页面，但是端口改成7681，例如 http://192.168.100.1:7681 ，输入用户名密码登录之后就可以执行命令了。

8389899, 2023-07-17:
可以用SSH连接，我马上试下，谢谢大佬！

8389899, 2023-07-17:
运行这个命令reset_rom_pkgs.run，系统配置会丢失吗？

8389899, 2023-07-18:
大佬，早上好！
昨天按您说的运行了reset_rom_pkgs.run，并重启系统，故障还是依旧进不了luci，提示出错还是跟原来一样，还有其它办法吗？

jjm2473, 2023-07-18:
执行这个命令试试 `is-opkg install luci-lib-iform`

8389899, 2023-07-18:
好的，谢谢大佬，晚上回去试下。

8389899, 2023-07-19:
按大佬说的方法，终于搞定了，谢谢大佬！