istoreos/istoreos #841: 打开istroe模块显示500 内部服务器错误

namemafan, 2023-06-19:
设备型号x86 Generic固件版本iStoreOS 21.02.3 2023060914，之前版本正常，升级后就打不开istroe模块了，其他模块正常，使用也正常，错误及日志如下：
错误：500 内部服务器错误
对不起，服务器遇到未知错误。
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'store/main'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:74: Failed to load template 'tasks/embed'.
Error while parsing template '/usr/lib/lua/luci/view/tasks/embed.htm':
No such file or directory
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:74: in function '__init__'
	/usr/lib/lua/luci/util.lua:65: in function 'Template'
	/usr/lib/lua/luci/dispatcher.lua:766: in function 'include'
	[string "/usr/lib/lua/luci/view/store/main.htm"]:3: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 
	(tail call): ?
	/usr/lib/lua/luci/controller/store.lua:134: in function 
日志：Mon Jun 19 11:26:55 2023 daemon.err uhttpd[6704]: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'store/main'.
Mon Jun 19 11:26:55 2023 daemon.err uhttpd[6704]: A runtime error occurred: /usr/lib/lua/luci/template.lua:74: Failed to load template 'tasks/embed'.
Mon Jun 19 11:26:55 2023 daemon.err uhttpd[6704]: Error while parsing template '/usr/lib/lua/luci/view/tasks/embed.htm':
Mon Jun 19 11:26:55 2023 daemon.err uhttpd[6704]: No such file or directory
Mon Jun 19 11:26:55 2023 daemon.err uhttpd[6704]: stack traceback:
Mon Jun 19 11:26:55 2023 daemon.err uhttpd[6704]: 	[C]: in function 'error'
Mon Jun 19 11:26:55 2023 daemon.err uhttpd[6704]: 	/usr/lib/lua/luci/template.lua:74: in function '__init__'
Mon Jun 19 11:26:55 2023 daemon.err uhttpd[6704]: 	/usr/lib/lua/luci/util.lua:65: in function 'Template'
Mon Jun 19 11:26:55 2023 daemon.err uhttpd[6704]: 	/usr/lib/lua/luci/dispatcher.lua:766: in function 'include'
Mon Jun 19 11:26:55 2023 daemon.err uhttpd[6704]: 	[string "/usr/lib/lua/luci/view/store/main.htm"]:3: in main chunk
Mon Jun 19 11:26:55 2023 daemon.err uhttpd[6704]: stack traceback:
Mon Jun 19 11:26:55 2023 daemon.err uhttpd[6704]: 	[C]: in function 'error'
Mon Jun 19 11:26:55 2023 daemon.err uhttpd[6704]: 	/usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85>
Mon Jun 19 11:26:55 2023 daemon.err uhttpd[6704]: 	(tail call): ?
Mon Jun 19 11:26:55 2023 daemon.err uhttpd[6704]: 	/usr/lib/lua/luci/controller/store.lua:134: in function </usr/lib/lua/luci/controller/store.lua:122>

namemafan, 2023-06-19:
[image]
[image]

jjm2473, 2023-06-19:
估计是文件损坏了，确认自己设备的根目录还有足够空间。 按这个教程可以重新安装或者修复istore  https://github.com/linkease/istore#%E5%AE%89%E8%A3%85-istore-%E5%88%B0-openwrt-%E5%AE%98%E6%96%B9%E5%9B%BA%E4%BB%B6

closed 2023-06-19 by jjm2473