istoreos/istoreos #847: socat做ipv6转换为ipv4端口转发访问不了，之前用的没问题

darcy212, 2023-06-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：socat做ipv6转换为ipv4端口转发访问不了，之前使用正常，防火墙端口也是打开的

### (2) 路由器型号和固件版本

A：x86工控机，固件版本 iStoreOS 21.02.3 2023060914

### (3) 详细日志和/或截图

A：使用ipv6+端口，一直跳转失败

darcy212, 2023-06-24:
> 反馈bug/问题模板，提建议请删除
>
> ## 1.关于你要提交的问题
> 为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue； 注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；
>
> Q：是否用关键词搜索了issue? (使用 "x" 选择)
>
> * [ ]  没有类似的issue
>
> ## 2. 详细叙述
> ### (1) 具体问题 [...]

closed 2023-06-24 by darcy212

BL1ACH, 2024-04-02:
看看上面启动了吗？看看防火墙-区域-wan口-入站流量是接受吗？丢弃或者拒绝是不通的。