istoreos/istoreos #848: 希望能添加airplay连接插件

jasonhuizhang, 2023-06-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：安装Shairplay插件，并开启服务。在iPhone中看不到此airplay服务。另外，在安装包中搜索不到airplay2插件。希望能提供airplay服务。谢谢

### (2) 路由器型号和固件版本

A：iStore [v0.1.13-3]，R4S

### (3) 详细日志和/或截图

A：

DaMccRee, 2023-06-25:
我也希望有

jjm2473, 2023-07-07:
哪里有airplay2？

jasonhuizhang, 2023-07-07:
之前用骷髅头固件自带有（https://github.com/DHDAXCW/NanoPi-R4S-R4SE）

jasonhuizhang, 2023-07-07:
> 哪里有airplay2？

之前用骷髅头固件里面有（https://github.com/DHDAXCW/NanoPi-R4S-R4SE）

jasonhuizhang, 2023-07-08:
> OK，应该是这个 https://github.com/coolsnowwolf/luci/blob/master/applications/luci-app-airplay2/Makefile

能加载到应用商店吗？不会编译🥹

jjm2473, 2023-07-12:
istore已经上架了此插件

closed 2023-07-12 by jjm2473