istoreos/istoreos #863: 升级后无法打开Web管理页面

wu0lei, 2023-07-08:
esxi安装istoreos，在web在线更新后，无法登录web配置页面，升级的版本号：22.03.4-2023070715-x86-64-squashfs-combined-efi.img.gz

jjm2473, 2023-07-08:
为什么不把日志发出来？既然是虚拟机，应该比物理机更容易抓日志或者截图吧

WYH2004-MC, 2023-07-08:
> 为什么不把日志发出来？既然是虚拟机，应该比物理机更容易抓日志或者截图吧

物理机升级后出现一样的情况，浏览器提示拒绝了连接，ssh可以连接，已经尝试重启uhttpd还是一样，请问要提供哪个日志，luci还是uhttpd

atomshadows, 2023-07-10:
我也遇到了，用SSH连接以后更新软件包和内核以后恢复了，你可以试试：

jjm2473, 2023-07-10:
别乱更新系统组件，反而容易出问题。

jjm2473, 2023-07-10:
@WYH2004-MC 系统日志。后台进不去可以用 logread 命令读取日志。或者自己打开浏览器的调试工具，看看有没有打印错误

jjm2473, 2023-07-10:
要么就恢复出厂设置

WYH2004-MC, 2023-07-11:
> @WYH2004-MC 系统日志。后台进不去可以用 logread 命令读取日志。或者自己打开浏览器的调试工具，看看有没有打印错误

解决了，我又开了一个Issues #865 里面有个人给出了解决方法

jjm2473, 2023-07-12:
如果之前使用nginx，现在打不开后台，那就升级nginx相关的包

jjm2473, 2023-07-12:
ssh或者终端（访问7681端口）执行 `opkg remove --force-removal-of-dependent-packages --autoremove nginx-util; opkg update ; opkg install luci-nginx`

closed 2023-07-24 by jjm2473