istoreos/istoreos #865: 更新22.03.4-2023070715-x86-64后,无法连接管理后台,提示拒绝了连接

WYH2004-MC, 2023-07-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
A：#863

## 2. 详细叙述
x86物理机，提供后台提供的在线更新，更新22.03.4-2023070715-x86-64后，出现无法连接管理后台的情况，浏览器提示拒绝了连接
ssh连接正常，尝试过重启luci和uhttpd服务，无果

### (1) 具体问题
无法连接Web管理后台

### (2) 路由器型号和固件版本
物理x86软路由(自己组的)
固件版本：istoreos-22.03.4-2023070715-x86-64

### (3) 详细日志和/或截图
（不知道要提供哪些日志，需要则后续补充）

WYH2004-MC, 2023-07-08:
系统日志
```

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 iStoreOS 22.03.4, 2023070715
 -----------------------------------------------------
root@iStoreOS:~# logread
Sat Jul  8 12:09:34 2023 authpriv.info dropbear[9180]: Child connection from 192.168.2.229:12302
Sat Jul  8 12:09:34 2023 daemon.err nlbwmon[9199]: Error loading shared library libubox.so.20210516: No such file or directory (needed by /usr/sbin/nlbwmon)
Sat Jul  8 12:09:34 2023 daemon.err nlbwmon[9199]: Error relocating /usr/sbin/nlbwmon: uloop_done: symbol not found
Sat Jul  8 12:09:34 2023 daemon.err nlbwmon[9199]: Error relocating /usr/sbin/nlbwmon: uloop_fd_add: symbol not found
Sat Jul  8 12:09:34 2023 daemon.err nlbwmon[9199]: Error relocating /usr/sbin/nlbwmon: avl_find: symbol not found
Sat Jul  8 12:09:34 2023 daemon.debug ModemManager[6759]: hotplug: cached event found: action=add, name=tty60, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty60
Sat Jul  8 12:09:34 2023 daemon.err nlbwmon[9199]: Error relocating /usr/sbin/nlbwmon: avl_delete: symbol not found
Sat Jul  8 12:09:34 2023 daemon.err nlbwmon[9199]: Error relocating /usr/sbin/nlbwmon: avl_init: symbol not found
Sat Jul  8 12:09:34 2023 daemon.err nlbwmon[9199]: Error relocating /usr/sbin/nlbwmon: uloop_timeout_cancel: symbol not found
Sat Jul  8 12:09:34 2023 daemon.err nlbwmon[9199]: Error relocating /usr/sbin/nlbwmon: __calloc_a: symbol not found
Sat Jul  8 12:09:34 2023 daemon.err nlbwmon[9199]: Error relocating /usr/sbin/nlbwmon: avl_insert: symbol not found
Sat Jul  8 12:09:34 2023 daemon.err nlbwmon[9199]: Error relocating /usr/sbin/nlbwmon: uloop_run_timeout: symbol not found
Sat Jul  8 12:09:34 2023 daemon.err nlbwmon[9199]: Error relocating /usr/sbin/nlbwmon: uloop_fd_delete: symbol not found
Sat Jul  8 12:09:34 2023 daemon.err nlbwmon[9199]: Error relocating /usr/sbin/nlbwmon: usock: symbol not found
Sat Jul  8 12:09:34 2023 daemon.err nlbwmon[9199]: Error relocating /usr/sbin/nlbwmon: uloop_timeout_set: symbol not found
Sat Jul  8 12:09:34 2023 daemon.err nlbwmon[9199]: Error relocating /usr/sbin/nlbwmon: uloop_init: symbol not found
Sat Jul  8 12:09:34 2023 user.notice nlbwmon: Reloading nlbwmon due to ifup of loopback (lo)
Sat Jul  8 12:09:34 2023 daemon.debug ModemManager[6759]: hotplug: event reported: action=add, name=tty60, subsystem=tty
Sat Jul  8 12:09:34 2023 daemon.err odhcpd[5545]: Failed to send to ff02::1%lan@br-lan (Address not available)
Sat Jul  8 12:09:34 2023 daemon.err ModemManager[6759]: hotplug: parent device sysfspath not found
Sat Jul  8 12:09:34 2023 daemon.debug ModemManager[6759]: hotplug: cached event found: action=add, name=tty22, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty22
Sat Jul  8 12:09:34 2023 daemon.debug ModemManager[6759]: hotplug: event reported: action=add, name=tty22, subsystem=tty
Sat Jul  8 12:09:34 2023 daemon.err ModemManager[6759]: hotplug: parent device sysfspath not found
Sat Jul  8 12:09:34 2023 daemon.debug ModemManager[6759]: hotplug: cached event found: action=add, name=tty6, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty6
Sat Jul  8 12:09:34 2023 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Sat Jul  8 12:09:34 2023 daemon.debug ModemManager[6759]: hotplug: event reported: action=add, name=tty6, subsystem=tty
Sat Jul  8 12:09:34 2023 authpriv.notice dropbear[9180]: Password auth succeeded for 'root' from 192.168.2.229:12302
Sat Jul  8 12:09:34 2023 daemon.err ModemManager[6759]: hotplug: parent device sysfspath not found
Sat Jul  8 12:09:34 2023 authpriv.info dropbear[9409]: Child connection from 192.168.2.229:12305
Sat Jul  8 12:09:34 2023 daemon.debug ModemManager[6759]: hotplug: cached event found: action=add, name=tty50, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty50
Sat Jul  8 12:09:34 2023 daemon.info avahi-daemon[6695]: Joining mDNS multicast group on interface br-lan.IPv6 with address fe80::2e0:4cff:fe1f:e2e.
Sat Jul  8 12:09:34 2023 daemon.info avahi-daemon[6695]: New relevant interface br-lan.IPv6 for mDNS.
Sat Jul  8 12:09:34 2023 daemon.info avahi-daemon[6695]: Registering new address record for fe80::2e0:4cff:fe1f:e2e on br-lan.*.
Sat Jul  8 12:09:34 2023 daemon.debug ModemManager[6759]: hotplug: event reported: action=add, name=tty50, subsystem=tty
Sat Jul  8 12:09:34 2023 daemon.err ModemManager[6759]: hotplug: parent device sysfspath not found
Sat Jul  8 12:09:34 2023 daemon.debug ModemManager[6759]: hotplug: cached event found: action=add, name=tty12, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty12
Sat Jul  8 12:09:34 2023 daemon.debug ModemManager[6759]: hotplug: event reported: action=add, name=tty12, subsystem=tty
Sat Jul  8 12:09:34 2023 daemon.err ModemManager[6759]: hotplug: parent device sysfspath not found
Sat Jul  8 12:09:34 2023 daemon.debug ModemManager[6759]: hotplug: cached event found: action=add, name=tty40, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty40
Sat Jul  8 12:09:34 2023 daemon.debug ModemManager[6759]: hotplug: event reported: action=add, name=tty40, subsystem=tty
Sat Jul  8 12:09:34 2023 daemon.err ModemManager[6759]: hotplug: parent device sysfspath not found
[... cut ...]
Sat Jul  8 12:09:40 2023 daemon.notice procd: /etc/rc.d/S99dockerd:  * Running script '/etc/zerotier.start'
Sat Jul  8 12:09:40 2023 daemon.notice procd: /etc/rc.d/S99dockerd:  * Running script '/var/etc/passwall.include'
Sat Jul  8 12:09:40 2023 daemon.notice procd: /etc/rc.d/S99dockerd:    ! Skipping due to path error: No such file or directory
Sat Jul  8 12:09:40 2023 daemon.notice procd: /etc/rc.d/S99dockerd:  * Running script '/var/etc/passwall_server.include'
Sat Jul  8 12:09:40 2023 daemon.notice procd: /etc/rc.d/S99dockerd:    ! Skipping due to path error: No such file or directory
```

willenwu21, 2023-07-08:
我也是，升级完这个版本无法连接Web管理后台，不过ssh是可以的。因为早把uhttpd换成了nginx，在重启下nginx时发现提示缺少libubus.so.20210630和libubox.so.20210516，然后去找这两个ipk安装完重启nginx就正常了

WYH2004-MC, 2023-07-08:
> 我也是，升级完这个版本无法连接Web管理后台，不过ssh是可以的。因为早把uhttpd换成了nginx，在重启下nginx时发现提示缺少libubus.so.20210630和libubox.so.20210516，然后去找这两个ipk安装完重启nginx就正常了

woc,牛的，我装完这两个包之后重启设备，Web就好了

wu0lei, 2023-07-08:
实验了下 确实有效

ANGELMARS, 2023-07-11:
我也是虚拟机，请问怎么用命令行打开ssh呀，web端进不去

WYH2004-MC, 2023-07-11:
> 我也是虚拟机，请问怎么用命令行打开ssh呀，web端进不去

你都能直接虚拟机命令行操作了，为什么还要用ssh

jjm2473, 2023-07-12:
如果是依赖问题，升级nginx相关的包就行了

jjm2473, 2023-07-12:
ssh或者终端（访问7681端口）执行 `opkg remove --force-removal-of-dependent-packages --autoremove nginx-util; opkg update ; opkg install luci-nginx`

jjm2473, 2023-07-12:
@ANGELMARS 浏览器打开7681端口就行了，例如之前你访问后台是 http://192.168.100.1/ ，改成 http://192.168.100.1:7681/ 就能打开终端。如果终端打不开，再考虑ssh

jjm2473, 2023-07-12:
重复issue，先关了

closed 2023-07-12 by jjm2473