istoreos/istoreos #866: R4S更新后 网络 - 接口 打开报错

byteinfo, 2023-07-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
网络 - 接口 打开报错
A：

### (2) 路由器型号和固件版本

FriendlyElec NanoPi R4S
iStoreOS 22.03.4 2023070715

### (3) 详细日志和/或截图

[image]

huangzulin, 2023-07-08:
我的更新后上不了网，重新刷最新的只有pwr灯亮，只能降级

KeplerSSSSS, 2023-07-09:
同样的情况,接口报错  不能上网

kokojacket, 2023-07-10:
问题+1，接口打不开报错，上不了网

Azhc, 2023-07-10:
我的也是 == 不知道怎么搞了现在 但是更神奇的是我外网通过ipv6还可以正常访问，就很奇怪

KeplerSSSSS, 2023-07-10:
重新安装之前的版本能上网了   但是依然打不开接口

balainit, 2023-07-10:
今天更新到最新版本后，遇到两个问题。
第一个是 USB 硬盘认不到了，不管怎么重启都扫描不到硬盘，重新拔插USB就可以了。
第二个是今天宕机了三次。上个版本的固件没宕机过。

kokojacket, 2023-07-11:
> 设备 r4s ，版本 iStoreOS 22.03.4 2023070715
> 
> 今天更新到最新版本后，遇到两个问题。
> 
> 第一个是 USB 硬盘认不到了，不管怎么重启都扫描不到硬盘，重新拔插USB就可以了。 第二个是今天宕机了三次。上个版本的固件没宕机过。
> 
> 我的接口这些一切正常的，就是宕机太频繁了。几十分钟宕机一次，受不了。
> 
> 怎么回退到上个版本？一堆东西，不想重新刷机，捣鼓了。

我是下老版本固件直接刷回去的，备份升级那一栏里有刷写固件，可以保配置刷写

jjm2473, 2023-07-12:
目前没有r4s可以测试，你们试试恢复出厂设置还有没有问题

Azhc, 2023-07-12:
刚好最近想换PVE，直接换了PVE，没用istore了，不折腾了 ==

byteinfo, 2023-07-12:
> 目前没有r4s可以测试，你们试试恢复出厂设置还有没有问题

配置比较多，没舍得重置，目测是前端的js文件缺失造成UI无法完成加载
[image]

byteinfo, 2023-07-12:
> > 目前没有r4s可以测试，你们试试恢复出厂设置还有没有问题
> 
> 配置比较多，没舍得重置，目测是前端的js文件缺失造成UI无法完成加载 [image]

我的 接口 界面可以正常打开了，感觉是浏览器缓存的js文件造成的

jjm2473, 2023-07-13:
靠，不是吧，浏览器缓存问题？js都是有版本号的，应该不会出现这种问题，是不是你之前更新过luci相关的组件？

closed 2023-07-13 by jjm2473

Azhc, 2023-07-13:
不像是js的问题 我后面无痕模式打开也是不行的。。。

jjm2473, 2023-07-13:
估计就是以前自己手动升级了luci组件导致的，istore那边安装systools，然后执行“修复系统组件”

jjm2473, 2023-07-13:
要么就恢复出厂设置

jjm2473, 2023-07-13:
那几个404错误是正常的，js会尝试加载所有可能的协议。最后一个 `L.naturalCompare` 错误是不正常的，肯定是因为安装了老版本luci导致的。

jjm2473, 2023-07-14:
那就用我说的方式就能修复