istoreos/istoreos #868: iStoreOS 22.03.4 2023070715 x86版本，x550网卡无法协商2.5G速率

MagicCheshireMeow, 2023-07-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
 iStoreOS 22.03.4 2023070715 x86版本，x550网卡无法协商2.5G速率，6.9版本的时候可以正常协商

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A： iStoreOS 22.03.4 2023070715 x86版本，x550网卡无法协商2.5G速率，6.9版本的时候可以正常协商。x550本身支持100/1000/2500/5000/10000五速率，且连接的设备本身也支持2.5G速率

### (2) 路由器型号和固件版本

A：x86-64-efi，主板EPYCD8，CPU为EPYC 7282，x550网卡为PCIE网卡。
      固件版本为iStoreOS 22.03.4 2023070715

### (3) 详细日志和/或截图

A：
root@AIOpenWrt:~# ethtool eth5
Settings for eth5:
        Supported ports: [ TP ]
        Supported link modes:   100baseT/Full
                                1000baseT/Full
                                10000baseT/Full
                                2500baseT/Full
                                5000baseT/Full
        Supported pause frame use: Symmetric
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  100baseT/Full
                                1000baseT/Full
                                10000baseT/Full
        Advertised pause frame use: Symmetric
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: on
        MDI-X: Unknown
        Supports Wake-on: d
        Wake-on: d
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: yes
root@AIOpenWrt:~# ethtool -s eth5 speed 2500 duplex full autoneg off
Cannot set new settings: Invalid argument
  not setting speed
  not setting duplex
  not setting autoneg

MagicCheshireMeow, 2023-07-11:
> 

root@AIOpenWrt:~# ethtool -i eth5
driver: ixgbe
version: 5.10.176
firmware-version: 0x80000aef, 1.1876.0
expansion-rom-version:
bus-info: 0000:47:00.1
supports-statistics: yes
supports-test: yes
supports-eeprom-access: yes
supports-register-dump: yes
supports-priv-flags: yes

jjm2473, 2023-07-11:
我看代码没有修改ixgbe驱动，可能是linux内核就已经限制了能选择的速率。你可以刷openwrt官方的22.03.4，看看测试结果是不是一样，如果结果一样，那只能让openwrt甚至linux那边修复了。

jjm2473, 2023-07-11:
https://github.com/openwrt/openwrt/issues/10459

MagicCheshireMeow, 2023-07-11:
> [openwrt/openwrt#10459](https://github.com/openwrt/openwrt/issues/10459)

感谢，那就等内核更新修复吧……

MagicCheshireMeow, 2023-07-11:
ethtool -s <ifname> advertise 0x1800000001028
这个命令实测可以开启所有支持的速率模式

closed 2023-07-11 by MagicCheshireMeow

jjm2473, 2023-07-11:
你可以在后台打开“系统”-“启动项”-“本地启动脚本”，把你的命令写到`exit 0`之前，这样开机时就会自动执行了

ChingyimLee, 2024-09-03:
> 你可以在后台打开“系统”-“启动项”-“本地启动脚本”，把你的命令写到`exit 0`之前，这样开机时就会自动执行了

大佬，这个命令怎么添加，前面要加sudo么？我直接添加在exit 0 前，重启发现没用。

jjm2473, 2024-09-13:
不需要sudo，默认都是root用户

akispho, 2025-05-24:
> > 你可以在后台打开“系统”-“启动项”-“本地启动脚本”，把你的命令写到`exit 0`之前，这样开机时就会自动执行了
> 
> 大佬，这个命令怎么添加，前面要加sudo么？我直接添加在exit 0 前，重启发现没用。

你要先确定你的网口 然后把<ifname>改成对应的

jjm2473, 2025-05-26:
这个issue已经好几年前的了