istoreos/istoreos #869: r5s更新22.03.4后无法拨号上网，附系统日志

maxguy66, 2023-07-11:
## 1.关于你要提交的问题
r5s在线更新22.03.4无法拨号，线刷后也无法拨号，刷回21系统可正常使用

系统日志：
Tue Jul 11 02:51:06 2023 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-mwan3.d: No such file or directory
Tue Jul 11 02:51:06 2023 daemon.crit dnsmasq[1]: FAILED to start up
Tue Jul 11 02:51:06 2023 daemon.info procd: - init complete -
Tue Jul 11 02:51:08 2023 daemon.warn pppd[11103]: Timeout waiting for PADO packets
Tue Jul 11 02:51:08 2023 daemon.err pppd[11103]: Unable to complete PPPoE Discovery
Tue Jul 11 02:51:08 2023 daemon.info pppd[11103]: Exit.
Tue Jul 11 02:51:08 2023 daemon.notice netifd: Interface 'wan' is now down
Tue Jul 11 02:51:08 2023 daemon.notice netifd: Interface 'wan' is setting up now
Tue Jul 11 02:51:08 2023 daemon.err insmod: module is already loaded - pppox
Tue Jul 11 02:51:08 2023 daemon.err insmod: module is already loaded - pppoe
Tue Jul 11 02:51:08 2023 daemon.info pppd[16421]: Plugin pppoe.so loaded.
Tue Jul 11 02:51:08 2023 daemon.info pppd[16421]: PPPoE plugin from pppd 2.4.9
Tue Jul 11 02:51:08 2023 daemon.notice pppd[16421]: pppd 2.4.9 started by root, uid 0
Tue Jul 11 02:51:09 2023 daemon.info samba4-server: io_uring module found, enabling VFS io_uring. (also needs Kernel 5.4+ Support)
Tue Jul 11 02:51:09 2023 daemon.info samba4-server: io_uring module found, enabling VFS io_uring. (also needs Kernel 5.4+ Support)
Tue Jul 11 02:51:09 2023 daemon.info samba4-server: io_uring module found, enabling VFS io_uring. (also needs Kernel 5.4+ Support)
Tue Jul 11 02:51:11 2023 daemon.crit dnsmasq[1]: cannot access directory /var/etc/dnsmasq-mwan3.d: No such file or directory
Tue Jul 11 02:51:11 2023 daemon.crit dnsmasq[1]: FAILED to start up
Tue Jul 11 02:51:11 2023 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash
Tue Jul 11 02:51:11 2023 daemon.notice ddnsto[12384]: dial ipv4 failed ret=-102 retry=4 remote=tunnel.tocmcc.cn:4445
Tue Jul 11 02:51:20 2023 daemon.notice ddnsto[12384]: dial ipv4 failed ret=-102 retry=5 remote=tunnel.tocmcc.cn:4445
Tue Jul 11 02:51:23 2023 daemon.warn pppd[16421]: Timeout waiting for PADO packets
Tue Jul 11 02:51:23 2023 daemon.err pppd[16421]: Unable to complete PPPoE Discovery
Tue Jul 11 02:51:23 2023 daemon.info pppd[16421]: Exit.
Tue Jul 11 02:51:23 2023 daemon.notice netifd: Interface 'wan' is now down
Tue Jul 11 02:51:23 2023 daemon.notice netifd: Interface 'wan' is setting up now
Tue Jul 11 02:51:23 2023 daemon.err insmod: module is already loaded - pppox
Tue Jul 11 02:51:23 2023 daemon.err insmod: module is already loaded - pppoe
Tue Jul 11 02:51:23 2023 daemon.info pppd[17233]: Plugin pppoe.so loaded.
Tue Jul 11 02:51:23 2023 daemon.info pppd[17233]: PPPoE plugin from pppd 2.4.9
Tue Jul 11 02:51:23 2023 daemon.notice pppd[17233]: pppd 2.4.9 started by root, uid 0
Tue Jul 11 02:51:24 2023 daemon.info samba4-server: io_uring module found, enabling VFS io_uring. (also needs Kernel 5.4+ Support)
Tue Jul 11 02:51:24 2023 daemon.info samba4-server: io_uring module found, enabling VFS io_uring. (also needs Kernel 5.4+ Support)
Tue Jul 11 02:51:24 2023 daemon.info samba4-server: io_uring module found, enabling VFS io_uring. (also needs Kernel 5.4+ Support)
Tue Jul 11 02:51:26 2023 daemon.err nmbd[11986]: [2023/07/11 02:51:26.806303,  0] ../../source3/nmbd/nmbd_become_lmb.c:398(become_local_master_stage2)
Tue Jul 11 02:51:26 2023 daemon.err nmbd[11986]:   *****
Tue Jul 11 02:51:26 2023 daemon.err nmbd[11986]:
Tue Jul 11 02:51:26 2023 daemon.err nmbd[11986]:   Samba name server ISTOREOS is now a local master browser for workgroup WORKGROUP on subnet 192.168.100.1
Tue Jul 11 02:51:26 2023 daemon.err nmbd[11986]:
Tue Jul 11 02:51:26 2023 daemon.err nmbd[11986]:   *****
Tue Jul 11 02:51:31 2023 daemon.notice ddnsto[12384]: dial ipv4 failed ret=-102 retry=6 remote=tunnel.tocmcc.cn:4445
Tue Jul 11 02:51:38 2023 daemon.warn pppd[17233]: Timeout waiting for PADO packets
Tue Jul 11 02:51:38 2023 daemon.err pppd[17233]: Unable to complete PPPoE Discovery
Tue Jul 11 02:51:38 2023 daemon.info pppd[17233]: Exit.
Tue Jul 11 02:51:38 2023 daemon.notice netifd: Interface 'wan' is now down
Tue Jul 11 02:51:38 2023 daemon.notice netifd: Interface 'wan' is setting up now
Tue Jul 11 02:51:38 2023 daemon.err insmod: module is already loaded - pppox
Tue Jul 11 02:51:38 2023 daemon.err insmod: module is already loaded - pppoe
Tue Jul 11 02:51:38 2023 daemon.info pppd[17726]: Plugin pppoe.so loaded.
Tue Jul 11 02:51:38 2023 daemon.info pppd[17726]: PPPoE plugin from pppd 2.4.9
Tue Jul 11 02:51:38 2023 daemon.notice pppd[17726]: pppd 2.4.9 started by root, uid 0
Tue Jul 11 02:51:40 2023 daemon.info samba4-server: io_uring module found, enabling VFS io_uring. (also needs Kernel 5.4+ Support)
Tue Jul 11 02:51:40 2023 daemon.info samba4-server: io_uring module found, enabling VFS io_uring. (also needs Kernel 5.4+ Support)
Tue Jul 11 02:51:40 2023 daemon.info samba4-server: io_uring module found, enabling VFS io_uring. (also needs Kernel 5.4+ Support)
Tue Jul 11 02:51:44 2023 daemon.notice ddnsto[12384]: dial ipv4 failed ret=-102 retry=7 remote=tunnel.tocmcc.cn:4445
Tue Jul 11 02:51:53 2023 daemon.warn pppd[17726]: Timeout waiting for PADO packets
Tue Jul 11 02:51:53 2023 daemon.err pppd[17726]: Unable to complete PPPoE Discovery
Tue Jul 11 02:51:53 2023 daemon.info pppd[17726]: Exit.
Tue Jul 11 02:51:53 2023 daemon.notice netifd: Interface 'wan' is now down
Tue Jul 11 02:51:53 2023 daemon.notice netifd: Interface 'wan' is setting up now
Tue Jul 11 02:51:53 2023 daemon.err insmod: module is already loaded - pppox
Tue Jul 11 02:51:53 2023 daemon.err insmod: module is already loaded - pppoe
Tue Jul 11 02:51:53 2023 daemon.info pppd[18219]: Plugin pppoe.so loaded.
Tue Jul 11 02:51:53 2023 daemon.info pppd[18219]: PPPoE plugin from pppd 2.4.9
Tue Jul 11 02:51:54 2023 daemon.notice pppd[18219]: pppd 2.4.9 started by root, uid 0
Tue Jul 11 02:51:55 2023 daemon.info samba4-server: io_uring module found, enabling VFS io_uring. (also needs Kernel 5.4+ Support)
[... 211 lines cut ...]
Tue Jul 11 02:55:12 2023 daemon.notice netifd: Interface 'wan' is setting up now
Tue Jul 11 02:55:12 2023 daemon.err insmod: module is already loaded - pppox
Tue Jul 11 02:55:12 2023 daemon.err insmod: module is already loaded - pppoe
Tue Jul 11 02:55:12 2023 daemon.info pppd[28512]: Plugin pppoe.so loaded.
Tue Jul 11 02:55:12 2023 daemon.info pppd[28512]: PPPoE plugin from pppd 2.4.9
Tue Jul 11 02:55:12 2023 daemon.notice pppd[28512]: pppd 2.4.9 started by root, uid 0
Tue Jul 11 02:55:13 2023 daemon.info samba4-server: io_uring module found, enabling VFS io_uring. (also needs Kernel 5.4+ Support)
Tue Jul 11 02:55:13 2023 daemon.info samba4-server: io_uring module found, enabling VFS io_uring. (also needs Kernel 5.4+ Support)
Tue Jul 11 02:55:13 2023 daemon.info samba4-server: io_uring module found, enabling VFS io_uring. (also needs Kernel 5.4+ Support)
Tue Jul 11 02:55:27 2023 daemon.warn pppd[28512]: Timeout waiting for PADO packets
Tue Jul 11 02:55:27 2023 daemon.err pppd[28512]: Unable to complete PPPoE Discovery
Tue Jul 11 02:55:27 2023 daemon.info pppd[28512]: Exit.
Tue Jul 11 02:55:27 2023 daemon.notice netifd: Interface 'wan' is now down
Tue Jul 11 02:55:27 2023 daemon.notice netifd: Interface 'wan' is setting up now
Tue Jul 11 02:55:27 2023 daemon.err insmod: module is already loaded - pppox
Tue Jul 11 02:55:27 2023 daemon.err insmod: module is already loaded - pppoe
Tue Jul 11 02:55:27 2023 daemon.info pppd[29462]: Plugin pppoe.so loaded.
Tue Jul 11 02:55:27 2023 daemon.info pppd[29462]: PPPoE plugin from pppd 2.4.9
Tue Jul 11 02:55:27 2023 daemon.notice pppd[29462]: pppd 2.4.9 started by root, uid 0
Tue Jul 11 02:55:28 2023 daemon.info samba4-server: io_uring module found, enabling VFS io_uring. (also needs Kernel 5.4+ Support)
Tue Jul 11 02:55:28 2023 daemon.info samba4-server: io_uring module found, enabling VFS io_uring. (also needs Kernel 5.4+ Support)
Tue Jul 11 02:55:28 2023 daemon.info samba4-server: io_uring module found, enabling VFS io_uring. (also needs Kernel 5.4+ Support)
Tue Jul 11 02:55:32 2023 daemon.notice ddnsto[29998]: dial ipv4 failed ret=-102 retry=1 remote=tunnel.tocmcc.cn:4445
Tue Jul 11 02:55:35 2023 daemon.notice ddnsto[29998]: dial ipv4 failed ret=-102 retry=2 remote=tunnel.tocmcc.cn:4445
Tue Jul 11 02:55:40 2023 daemon.notice ddnsto[29998]: dial ipv4 failed ret=-102 retry=3 remote=tunnel.tocmcc.cn:4445

jjm2473, 2023-07-12:
dnsmasq启动失败，看起来是mwan3导致的，卸载掉mwan3。
如果还有问题，就恢复出厂设置。

jjm2473, 2023-07-13:
那你试试不保留配置刷回21.02固件，然后手动配置拨号，看有没有问题

jjm2473, 2023-07-13:
我怀疑你是不是自己改过wan口却忘了

maxguy66, 2023-07-13:
刷回21固件后：
- 通过网络向导配置拨号可以正常上网
- 手动配置wan口拨号可以正常上网
- 导入旧的配置后可以正常上网

升级22不保留配置不能上网
- 通过firstboot重置后再通过网络向导配置拨号不能上网

wan口配置没有手动改过

jjm2473, 2023-07-13:
那就不知道咋回事了，也没其他人反馈。要么你就问问运营商，有没有过滤mac地址什么的

maxguy66, 2023-07-18:
@jjm2473 找到问题了，因为墙壁上的网口是百兆的，21版本的系统wan（默认关联etn0）口可以自适应百兆，22系统不会自适应，插上去协商速率还是千兆，所以无法拨号，手动设置一下接口，把wan口改成eth1，2.5G的eth1或者eth2可以自适应百兆，再拨号就可以正常用了

jjm2473, 2023-07-18:
r5s的wan口不能自适应百兆吗

maxguy66, 2023-07-18:
@jjm2473 是的，从22固件开始，wan不会自适应百兆了，插到百兆的网口就不能拨号了，可以临时把wan换成eth1，就可以正常用了

jjm2473, 2023-07-18:
我特意找了个只有百兆口的摄像头，连到r5s的wan口，没啥问题啊。

[image]

不过没测试拨号，现在已经很难找到只有百兆的设备来测试拨号了。

你把wan口接上百兆的设备，然后用`ethtool eth0`看看是不是协商到百兆了

maxguy66, 2023-07-18:
eth0设置成lan口状态下，插了一个百兆口的AP，确实可以协商出百兆，这个物理口设置成wan口插墙壁上的网口拨号就是协商出来千兆，拨号不成功，换成其他物理网口再插墙上就能协商百兆正常上网

closed 2023-07-24 by jjm2473