istoreos/istoreos #871: 关于友善r6s  22.03.4固件ipv6的问题

pili-book, 2023-07-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
感觉默认的ipv6设置不是很好，无法通过ipv6测试及迅雷ipv6测试，最近老是迅雷下载加速慢过老一会才能起来速度，怀疑是单ip限制了连接数（小白不是很清楚哪里改这东西），今天发现ipv6测试不通过，所以各种搜索瞎改了半天，希望固件默认的这个东西再调试下

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：首先固件首页不显示默认从电信获取的ipv6 DNS地址，自己瞎搜索调试吧防火墙入站转发全设置为接受，迅雷ipv6显示支持了，然后自己把电脑的ipv6 dns手动设置为网上找的一个地址，通过了http://test-ipv6.com/的检测（有时候通过有时候通不过，不稳定？）

### (2) 路由器型号和固件版本

A：友善R6s 最新固件0715

### (3) 详细日志和/或截图

A：

pili-book, 2023-07-11:
顺便希望大神告知单ip连接数限制哪里修改还是说本来就是最大值了，我家里就我这一台电脑迅雷下载，希望设置高点

jjm2473, 2023-07-12:
搜索下issue和discussions，ipv6的问题早就回答过了。

jjm2473, 2023-07-12:
https://github.com/istoreos/istoreos/discussions/646

closed 2023-07-12 by jjm2473

pili-book, 2023-07-12:
> #646
ipv6地址应该没问题，但是默认关闭了DNS的ipv6解析，以免网页加载慢，如果要开启ipv6解析，可以到“DHCP/DNS”的高级设置配置，第一个选项就是
这个答案吗？我的固件第一项是 不记录日志，没看懂到底是哪个选项 22.03固件

jjm2473, 2023-07-13:
搜一下ipv6啊

jjm2473, 2023-07-13:
连第几个选项还要说吗

pili-book, 2023-07-13:
> 连第几个选项还要说吗

搜了兄弟，就是没有对照的答案哦。。。。新固件和以前的不一样了，我自己手动加了两个dns地址，首页那里也不显示运营商那获取的ipv6 的dns地址，那个选项我这个固件上就没有的，顺便问问单ip连接数限制哪里修改（百度修改/etc/sysctl.conf这个文件。。但是新固件就没有）

jjm2473, 2023-07-13:
都说了，在“DHCP/DNS”的高级设置，这都找不到

jjm2473, 2023-07-13:
[image]

jjm2473, 2023-07-14:
dns跟ipv6没有太大关系，dns地址是ipv4的也能提供ipv6解析。关键是，让你关掉ipv6的过滤啊，都截图给你了

pili-book, 2023-07-14:
> dns跟ipv6没有太大关系，dns地址是ipv4的也能提供ipv6解析。关键是，让你关掉ipv6的过滤啊，都截图给你了

哦这个位置找见了，主要是以为能直接从运营商那获取IPV6 dns地址了，打电话给运营商好像不行，他们也闹不懂，多谢，知道这个选项干啥的了