istoreos/istoreos #872: 网络接口状态空

666dechaoge, 2023-07-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：网络接口状态空，网口配置页面空，但是接口以及网络一切正常

### (2) 路由器型号和固件版本

A：路由器R2S 固件iStoreOS 22.03.4 2023070715 / LuCI openwrt-22.03 branch git-23.119.80898-65ef406

### (3) 详细日志和/或截图

A：
[image]
[image]
[image]

jjm2473, 2023-07-13:
实测没有这个问题。

根据这里的第2条修复系统组件 https://github.com/istoreos/istoreos/wiki/FAQ

要么就恢复出厂设置

closed 2023-07-13 by jjm2473

666dechaoge, 2023-07-13:
恢复sys解决了，感谢

Vinke2, 2024-08-02:
+1 用系统修复工具解决了