istoreos/istoreos #873: 关于5.10rockchip uboot打包失败的解决方式（ImportError: No module named elftools.elf.elffile）

Beaverfffan, 2023-07-12:
其他源码已经在使用python3-pyelftools。istoreos使用的是rk官方make_fit_atf.py
需要切换到python2的pyelftools
sudo apt install python2 python-pip
pip2 install pyelftools
把python3的版本删删除

以下是错误出现的情况
./"arch/arm/mach-rockchip/make_fit_atf.py"
arch/arm/dts/rk3568-easepi.dtb > u-boot.its
Traceback (most recent call last):
File "./arch/arm/mach-rockchip/make_fit_atf.py", line 15, in
from elftools.elf.elffile import ELFFile
ImportError: No module named elftools.elf.elffile
make[4]: *** [Makefile:1018: u-boot.its] Error 1
make[4]: Leaving directory '/home/beaver/istoreos/build_dir/target-aarch64_cortex-a53_musl/u-boot-easepi-rk3568/u-boot-2017.09-20230602'
make[3]: *** [Makefile:84: /home/beaver/istoreos/build_dir/target-aarch64_cortex-a53_musl/u-boot-easepi-rk3568/u-boot-2017.09-20230602/.built] Error 2
make[3]: Leaving directory '/home/beaver/istoreos/package/boot/uboot-rk35xx'
time: package/boot/uboot-rk35xx/easepi-rk3568/compile#3.44#2.15#5.48
ERROR: package/boot/uboot-rk35xx failed to build (build variant: easepi-rk3568).
make[2]: *** [package/Makefile:116: package/boot/uboot-rk35xx/compile] Error 1
make[2]: Leaving directory '/home/beaver/istoreos'
make[1]: *** [package/Makefile:110: /home/beaver/istoreos/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/beaver/istoreos'
make: *** [/home/beaver/istoreos/include/toplevel.mk:234: world] Error 2

jjm2473, 2023-07-12:
👍

Beaverfffan, 2023-07-12:
and glib2.0 needs to install python3-distutils manually 

[meson-log.txt](https://github.com/istoreos/istoreos/files/12023902/meson-log.txt)

Beaverfffan, 2023-07-12:
istoreos的环境配置大概非常需要特别说明如何配置。是否可以出一个官方环境配置文档在read.me?
@jjm2473

Beaverfffan, 2023-07-12:
sudo apt update -y
sudo apt full-upgrade -y
sudo apt install ack antlr3 asciidoc autoconf automake autopoint binutils bison build-essential \
bzip2 ccache clang cmake cpio curl device-tree-compiler fastjar flex gawk \
gcc-multilib g++-multilib gcc gettext git gperf haveged help2man intltool \
libc6-dev-i386 libelf-dev libglib2.0-dev libgmp3-dev libltdl-dev libmpc-dev \
libmpfr-dev libncurses5-dev libncursesw5-dev libpython3-dev libreadline-dev \
libssl-dev libtool libncurses-dev libpython3-dev lrzsz mkisofs msmtp nano \
ninja-build p7zip p7zip-full patch pkgconf python2.7 python3 python3-distutils \
python3-setuptools qemu-utils rsync scons squashfs-tools subversion swig \
texinfo uglifyjs unzip upx-ucl vim wget xmlto xxd zlib1g-dev
使用rk uboot的特别要求
sudo apt install python2 python-pip
pip2 install pyelftools

jjm2473, 2023-07-12:
@Beaverfffan 除了rk的uboot以外，其他都是跟openwrt相同的编译环境，如果遇到问题，google下也能解决了

jjm2473, 2023-07-24:
最新 22.03 代码已经将 make_fit_atf.py 替换成 make_fit_atf.sh ，应该不会再出现这个问题

closed 2023-07-24 by jjm2473

dgyangxf2023, 2024-07-03:
最新22.03编译时仍然会提示。

AnyoeDove, 2024-07-06:
今天编译H28K时又出现了同样的报错...  `pip2 install pyelftools==0.27` 暂时解决 ....

jjm2473, 2024-07-17:
你们用的是22.03分支最新代码吗？单独编译uboot看看日志。编译命令`make V=s package/uboot-rk35xx/compile`