istoreos/istoreos #874: Transmission可以更新到4.0.3吗？

18309871333, 2023-07-12:
Transmission现在3.0的版本种子多了之后总重启， 重启后种子都消失掉了。

jjm2473, 2023-07-12:
transmission是openwrt官方仓库提供的，我们一般不会去升级，除非确定有严重bug

jjm2473, 2023-07-12:
如果你要升级，可以到这里手动下载ipk https://downloads.openwrt.org/releases/packages-23.05/

18309871333, 2023-07-12:
感谢！

scrymile, 2023-07-27:
现在就是有严重bug。一添加种子就无法连接了

18309871333, 2023-07-27:
现在全靠qB自己撑着几千个种子，可把它累坏了

jjm2473, 2023-07-27:
> > 现在就是有严重bug。一添加种子就无法连接了
> 
> 我也遇到了同样的问题，全新安装了一次iStoreos，安装完Tr，下载第一个种子马上无法连接到tr。

难道新版本就没问题？据我所知，tr在分配磁盘的时候会假死，等分配完以后就会恢复了，这跟版本应该无关。

18309871333, 2023-07-27:
> > > 现在就是有严重bug。一添加种子就无法连接了
> > 
> > 
> > 我也遇到了同样的问题，全新安装了一次iStoreos，安装完Tr，下载第一个种子马上无法连接到tr。
> 
> 难道新版本就没问题？据我所知，tr在分配磁盘的时候会假死，等分配完以后就会恢复了，这跟版本应该无关。

以前不这样，就最近才出现的

jjm2473, 2023-09-02:
我复现了“一添加种子就无法连接”的问题，好像只在x86_64会出现。

修复方案已经pr给openwrt了 https://github.com/openwrt/packages/pull/21996 。

在openwrt发布修复版本之前，如果想要手动修复，可以下载 [transmission-daemon.json](https://raw.githubusercontent.com/openwrt/packages/780b13ed20add5fa28dde654964fb516863eb781/net/transmission/files/transmission-daemon.json) 替换掉系统里面的`/etc/seccomp/transmission-daemon.json` 。

以上只针对固件版本 22.03.*，transmission 3.00

jjm2473, 2023-10-25:
openwrt已经合并并且发布ipk，如果还有问题就重新安装tr。

Artyaya, 2023-11-17:
> openwrt已经合并并且发布ipk，如果还有问题就重新安装tr。

重装也不行，修改/etc/seccomp/transmission-daemon.json后，有所改善，但还是会死，时间撑久一点而已。

你把transmission配置里的 队列暂停分钟数改到3分钟，添加十几个下载队列，就能很快发现，不到10分钟进程就死了。

jjm2473, 2023-11-19:
修改/etc/seccomp/transmission-daemon.json 不是因为进程卡死，而是因为seccomp，也就是安全沙箱，tr的所有系统调用都要写进这个json清单里面

jjm2473, 2023-11-19:
如果还有问题，就关掉seccomp功能