istoreos/istoreos #876: H68K更新到iStoreOS 22.03.4 2023070715，无法识别网口

CharlieVSAngel, 2023-07-13:
H68K更新到iStoreOS 22.03.4 2023070715后，四网口无法正常识别，智能识别到三个网口，而且网口顺序不对
[image]
[image]

litongda007, 2023-07-13:
同样问题可以稳定复现在我的H68K上。另外设备型号识别也有误，会被识别为H69K。

jjm2473, 2023-07-13:
因为h68k硬件有很多版本，有些硬件版本没法在软件上区分，我们也不会给一个型号分多个固件的，有问题就找厂商吧。

jjm2473, 2023-07-13:
早期版本的h68k无法与现在的h69k区分开，于是识别成h69k了，h69k就是三个网口的

CharlieVSAngel, 2023-07-13:
> 

为什么之前的版本就不会，升级到iStoreOS 22.03.4 2023070715后就出问题了，按照这个思路是不是可以解决？

CharlieVSAngel, 2023-07-13:
> 早期版本的h68k无法与现在的h69k区分开，于是识别成h69k了，h69k就是三个网口的

可不可以自己手动添加端口？

为什么之前的版本就不会，升级到iStoreOS 22.03.4 2023070715后就出问题了，按照这个思路是不是可以解决？

jjm2473, 2023-07-14:
因为之前的版本没有h69k啊

lan-zz, 2023-07-16:
我的H68K，首页的网络接口状态与侧边菜单点出来的网口配置里都看不见网口，点击保存并应于就出现“网络异常：500 response parse failed”报错 
[image]

jjm2473, 2023-07-17:
@lan-zz 1. 换浏览器，2. 恢复出厂设置。总能修复的

closed 2023-07-24 by jjm2473

BlackteaCN, 2025-06-13:
<转载>
1.在终端执行
cp /boot/rockchip1.dtb  /boot/rockchip10.dtb
2.重启
3.接口中发现新接口
4.将新接口手动添加到lan。
5.第二步也可以改为直接恢复出厂设置
6.本教程仅限能正常识别三口的机型使用，原本只能识别一口的只能恢复双千兆，仍然无法使用2.5g。
这些最老的机型需要刷固定型号的固件，无法使用istoreos自动判别。

jjm2473, 2025-06-13:
主要就是一些老的机器没有配置好ADC的电阻，导致无法跟其他版本区分开来，这样的机器应该很少。有能力的可以问厂商怎么改电阻，然后手动改。

手动替换dtb文件的话，升级固件以后又会变回来。

另外，如果熟悉uboot，可以进uboot的console，修改启动命令。uboot是不会被升级覆盖的。