istoreos/istoreos #879: 更新iStoreOS 22.03.4 2023070715后经常掉线

kinf520, 2023-07-15:
不清楚是哪里的问题，自从更新这边版本后长期掉线自动重连。
[image]
[image]
[image]

kinf520, 2023-07-16:
[image]
捕捉到一个最新断网时间17.29分

jjm2473, 2023-07-18:
发日志能不能直接复制文本呢，有什么必要截图

jjm2473, 2023-07-18:
[image]
确定不是有人在拔插网线？先是eth3，然后eth0

kinf520, 2023-07-18:
没人拔网线啊，而且每次断网都是随机的。有时候一天多，有时候几小时

jjm2473, 2023-07-24:
那是不是过热了

closed 2023-08-05 by jjm2473