istoreos/istoreos #883: 请求添加mosdns

draco-china, 2023-07-17:
## 1.关于你要提交的问题
手动安装 mosdns 太过繁琐
[luci-app-mosdns](https://github.com/sbwml/luci-app-mosdns)
目前手动安装已正常运行

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
手动安装需要先安装mosdns v2dat 才能继续安装luci-app-mosdns及汉化包

### (1) 具体问题
为什么不适用smart dns ?
A：相对而言 smart dns 手动配置麻烦，mosdns 自动分流

### (2) 路由器型号和固件版本

A：

### (3) 详细日志和/或截图

A：

Jarv1s0, 2023-07-19:
你好，请问我在安装过程中出现以下以来错误，不知道你是在哪里下载的libubox20210516 和v2dat。我的是R4S，下载的aarch64_generic，提示不兼容。
Unknown package 'luci-app-mosdns'.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libubox20210516 for jsonfilter
 * pkg_hash_check_unresolved: cannot find dependency v2dat for luci-app-mosdns
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-mosdns found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-mosdns.

jjm2473, 2023-07-28:
这里有离线安装包 https://github.com/AUK9527/Are-u-ok

closed 2023-07-28 by jjm2473