istoreos/istoreos #886: 升级后出现  curl: (48) Error ！

cn-jiangfei, 2023-07-17:
## 1.关于你要提交的问题
Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
前几天看到有本版更新就随手更新了一下，更新以后没有注意，今天发现检测更新或者安装应用时提示 curl: (48) Error！
### (1) 具体问题

A：检查和更新固件时提示 curl: (48) Error！
B：iStore安装应用时提示 curl: (48) Error！
C：使用 curl 命令时提示 curl: (48) Error！

### (2) 路由器型号和固件版本

A：EXSI 虚拟机  x86/64
B：固件版本	iStoreOS 22.03.4 2023070715 / LuCI istoreos-22.03 branch git-23.103.23275-0193674
C ：内核版本	5.10.176

### (3) 详细日志和/或截图

A：
[image]

[image]

[image]

cn-jiangfei, 2023-07-17:
已经解决，主要是因为   curl 和 libcurl 版本不匹配 导致的。
`root@Openwrt:~# curl -V
curl 7.88.1 (x86_64-openwrt-linux-gnu) libcurl/7.83.1 wolfSSL/5.5.3 nghttp2/1.44.0
Release-Date: 2023-02-20
Protocols: file ftp ftps http https mqtt
Features: alt-svc HSTS HTTP2 IPv6 Largefile SSL
WARNING: curl and libcurl versions do not match. Functionality may be affected.`

这时候更新一个  curl 和 libcurl 版本，让他们都是最新的：

jjm2473, 2023-07-18:
什么不匹配啊，就是因为你自己升级导致的啊

closed 2023-07-18 by jjm2473