istoreos/istoreos #888: 使用最新固件，过段时间DHCP服务不会自动分配导致无网络，需要重启设备或者手动设置DHCP才能解决

AsherPeople, 2023-07-17:
Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
使用最新固件，过段时间DHCP服务不会自动分配导致无网络，需要重启设备或者手动设置DHCP才能解决
### (1) 具体问题
A：
[image]

### (2) 路由器型号和固件版本
A： x86/64  英特尔® 赛扬® 处理器 J1900
B： 固件版本 iStoreOS 22.03.4 2023071416 / LuCI istoreos-22.03 branch git-23.103.23275-0193674
C：内核版本：5.10.176
### (3) 详细日志和/或截图
A：系统日志：
Tue Jul 18 06:24:51 2023 user.warn ddns-scripts[15384]: luo: Get registered/public IP for 'rowan-willa.asia' failed - retry 526/0 in 60 seconds
Tue Jul 18 06:25:18 2023 daemon.err /usr/bin/ss-local[5302]: remote_recv_cb_recv: Connection reset by peer
Tue Jul 18 06:25:18 2023 daemon.err /usr/bin/ss-local[5302]: remote_recv_cb_recv: Connection reset by peer
Tue Jul 18 06:25:19 2023 daemon.err /usr/bin/ss-local[5302]: remote_recv_cb_recv: Connection reset by peer
Tue Jul 18 06:25:29 2023 daemon.err /usr/bin/ss-local[6532]: remote_recv_cb_recv: Connection reset by peer
Tue Jul 18 06:25:51 2023 user.warn ddns-scripts[15384]: luo: NO valid IP found
Tue Jul 18 06:26:51 2023 user.warn ddns-scripts[15384]: luo: NO valid IP found
Tue Jul 18 06:29:52 2023 user.warn ddns-scripts[15384]: luo: NO valid IP found
Tue Jul 18 06:29:52 2023 user.warn ddns-scripts[15384]: luo: Get registered/public IP for 'rowan-willa.asia' failed - retry 531/0 in 60 seconds
Tue Jul 18 06:31:41 2023 kern.info kernel: [32622.940402] usb 1-2.3: USB disconnect, device number 12
Tue Jul 18 06:31:41 2023 kern.err kernel: [32622.941363] blk_update_request: I/O error, dev sde, sector 0 op 0x1:(WRITE) flags 0x800 phys_seg 0 prio class 0
Tue Jul 18 06:31:41 2023 kern.notice kernel: [32622.962570] sd 4:0:0:0: [sde] Synchronizing SCSI cache
Tue Jul 18 06:31:42 2023 kern.info kernel: [32623.639070] sd 4:0:0:0: [sde] Synchronize Cache(10) failed: Result: hostbyte=0x07 driverbyte=0x00
Tue Jul 18 06:31:47 2023 kern.info kernel: [32628.878861] usb 1-2.3: new full-speed USB device number 13 using xhci_hcd
Tue Jul 18 06:31:47 2023 kern.info kernel: [32629.054872] input: USB Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.3/1-2.3:1.0/0003:04D9:A0CD.0004/input/input13
Tue Jul 18 06:31:47 2023 kern.info kernel: [32629.119060] hid-generic 0003:04D9:A0CD.0004: input,hidraw0: USB HID v1.11 Keyboard [USB Keyboard] on usb-0000:00:14.0-2.3/input0
Tue Jul 18 06:31:47 2023 kern.info kernel: [32629.127015] input: USB Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.3/1-2.3:1.1/0003:04D9:A0CD.0005/input/input14
Tue Jul 18 06:31:47 2023 kern.info kernel: [32629.188695] input: USB Keyboard System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.3/1-2.3:1.1/0003:04D9:A0CD.0005/input/input15
Tue Jul 18 06:31:47 2023 kern.info kernel: [32629.188886] input: USB Keyboard Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.3/1-2.3:1.1/0003:04D9:A0CD.0005/input/input16
Tue Jul 18 06:31:47 2023 kern.info kernel: [32629.189052] hid-generic 0003:04D9:A0CD.0005: input,hiddev96,hidraw1: USB HID v1.11 Keyboard [USB Keyboard] on usb-0000:00:14.0-2.3/input1
Tue Jul 18 06:31:47 2023 kern.info kernel: [32629.192630] hid-generic 0003:04D9:A0CD.0006: hiddev97,hidraw2: USB HID v1.11 Device [USB Keyboard] on usb-0000:00:14.0-2.3/input2
Tue Jul 18 06:31:59 2023 daemon.notice netifd: Network device 'eth3' link is up
Tue Jul 18 06:31:59 2023 kern.info kernel: [32640.995799] tg3 0000:03:00.3 eth3: Link is up at 1000 Mbps, full duplex
Tue Jul 18 06:31:59 2023 kern.info kernel: [32640.995811] tg3 0000:03:00.3 eth3: EEE is disabled
Tue Jul 18 06:31:59 2023 kern.info kernel: [32640.995843] br-lan: port 3(eth3) entered blocking state
Tue Jul 18 06:31:59 2023 kern.info kernel: [32640.995850] br-lan: port 3(eth3) entered forwarding state
Tue Jul 18 06:32:00 2023 kern.info kernel: [32642.017600] tg3 0000:03:00.3 eth3: Link is down
Tue Jul 18 06:32:00 2023 daemon.notice netifd: Network device 'eth3' link is down
Tue Jul 18 06:32:00 2023 kern.info kernel: [32642.027932] br-lan: port 3(eth3) entered disabled state
Tue Jul 18 06:32:03 2023 daemon.notice netifd: Network device 'eth3' link is up
Tue Jul 18 06:32:03 2023 kern.info kernel: [32644.896689] tg3 0000:03:00.3 eth3: Link is up at 1000 Mbps, full duplex
Tue Jul 18 06:32:03 2023 kern.info kernel: [32644.896702] tg3 0000:03:00.3 eth3: EEE is disabled
Tue Jul 18 06:32:03 2023 kern.info kernel: [32644.896730] br-lan: port 3(eth3) entered blocking state
Tue Jul 18 06:32:03 2023 kern.info kernel: [32644.896737] br-lan: port 3(eth3) entered forwarding state
Tue Jul 18 06:32:52 2023 user.warn ddns-scripts[15384]: luo: NO valid IP found
Tue Jul 18 06:32:52 2023 user.warn ddns-scripts[15384]: luo: Get registered/public IP for 'rowan-willa.asia' failed - retry 534/0 in 60 seconds
Tue Jul 18 06:32:56 2023 daemon.notice netifd: Network device 'eth3' link is down
Tue Jul 18 06:32:56 2023 kern.info kernel: [32697.516103] tg3 0000:03:00.3 eth3: Link is down
Tue Jul 18 06:32:56 2023 kern.info kernel: [32697.516300] br-lan: port 3(eth3) entered disabled state
Tue Jul 18 06:32:59 2023 daemon.notice netifd: Network device 'eth3' link is up
Tue Jul 18 06:32:59 2023 kern.info kernel: [32700.470127] tg3 0000:03:00.3 eth3: Link is up at 1000 Mbps, full duplex
Tue Jul 18 06:32:59 2023 kern.info kernel: [32700.470136] tg3 0000:03:00.3 eth3: Flow control is on for TX and on for RX
Tue Jul 18 06:32:59 2023 kern.info kernel: [32700.470140] tg3 0000:03:00.3 eth3: EEE is enabled
Tue Jul 18 06:32:59 2023 kern.info kernel: [32700.470171] br-lan: port 3(eth3) entered blocking state
Tue Jul 18 06:32:59 2023 kern.info kernel: [32700.470178] br-lan: port 3(eth3) entered forwarding state
Tue Jul 18 06:33:52 2023 user.warn ddns-scripts[15384]: luo: NO valid IP found
Tue Jul 18 06:33:52 2023 user.warn ddns-scripts[15384]: luo: Get registered/public IP for 'rowan-willa.asia' failed - retry 535/0 in 60 seconds
Tue Jul 18 06:34:18 2023 user.info : luci: accepted login on / for root from 192.168.100.2
B：内核日志
[    0.101292] ENERGY_PERF_BIAS: Set to 'normal', was 'performance'
[    0.106318] cryptd: max_cpu_qlen set to 1000
[    0.107109] ACPI: Added _OSI(Module Device)
[    0.107119] ACPI: Added _OSI(Processor Device)
[    0.107125] ACPI: Added _OSI(3.0 _SCP Extensions)
[    0.107131] ACPI: Added _OSI(Processor Aggregator Device)
[... 786 lines cut ...]
[32640.995811] tg3 0000:03:00.3 eth3: EEE is disabled
[32640.995843] br-lan: port 3(eth3) entered blocking state
[32640.995850] br-lan: port 3(eth3) entered forwarding state
[32642.017600] tg3 0000:03:00.3 eth3: Link is down
[32642.027932] br-lan: port 3(eth3) entered disabled state
[32644.896689] tg3 0000:03:00.3 eth3: Link is up at 1000 Mbps, full duplex
[32644.896698] tg3 0000:03:00.3 eth3: Flow control is off for TX and off for RX
[32644.896702] tg3 0000:03:00.3 eth3: EEE is disabled
[32644.896730] br-lan: port 3(eth3) entered blocking state
[32644.896737] br-lan: port 3(eth3) entered forwarding state
[32697.516103] tg3 0000:03:00.3 eth3: Link is down
[32697.516300] br-lan: port 3(eth3) entered disabled state
[32700.470127] tg3 0000:03:00.3 eth3: Link is up at 1000 Mbps, full duplex
[32700.470136] tg3 0000:03:00.3 eth3: Flow control is on for TX and on for RX
[32700.470140] tg3 0000:03:00.3 eth3: EEE is enabled
[32700.470171] br-lan: port 3(eth3) entered blocking state
[32700.470178] br-lan: port 3(eth3) entered forwarding state
[34222.706678] tg3 0000:03:00.3 eth3: Link is down
[34222.706767] br-lan: port 3(eth3) entered disabled state
[34227.177872] tg3 0000:03:00.3 eth3: Link is up at 1000 Mbps, full duplex
[34227.177877] tg3 0000:03:00.3 eth3: Flow control is on for TX and on for RX
[34227.177880] tg3 0000:03:00.3 eth3: EEE is enabled
[34227.177907] br-lan: port 3(eth3) entered blocking state
[34227.177912] br-lan: port 3(eth3) entered forwarding state
C：[image]

AsherPeople, 2023-07-17:
[image]

jjm2473, 2023-07-18:
把翻墙插件卸载掉重启，之后再看看还有没有问题

maxsusu, 2023-07-20:
建议你把所有容器圈闭关闭测试一段时间看看，我觉得是容器的BUG导致这个问题，我们自己测试把容器停了就不会出现这个问题

closed 2023-07-24 by jjm2473