istoreos/istoreos #894: 公网DDNS访问软路由域名失败，直接访问IPv6地址可以

shixin-679, 2023-07-21:
固件版本：iStoreOS 22.03.4 2023071416 / LuCI istoreos-22.03 branch git-23.103.23275-0193674
故障现象：通过DDNS实现公网IPv6访问软路由，并且挂了SSL证书的，以前的版本是没问题的，但是近期使用域名访问失败，公网电脑有IPv6地址，使用域名解析出来的IPv6地址（DNS解析是没问题的，无论公网还是内网电脑）直接访问又可以。。。
截图：
[image]
[image]
[image]
[image]

shixin-679, 2023-07-21:
求助，实在排查不出来了

jjm2473, 2023-07-24:
自己看看ddns是不是正常啊，有没有绑定好域名，还有端口是不是正确？如果你用了端口转发，那要勾选NAT环回

closed 2023-07-24 by jjm2473