istoreos/istoreos #898: 公网IP变动出现断网情况。

kinf520, 2023-07-23:
发现最近的版本动态IP变更后都出现断网现象。
设备型号WIZARD MAGIC N95/N100/N200/N305
固件版本iStoreOS 22.03.4 2023071416
内核版本 5.10.176
系统日志：
Sun Jul 23 01:21:50 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 10.10.10.106 88:97:46:07:fa:e6
Sun Jul 23 01:32:24 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 10.10.10.136 24:a1:60:2a:d0:e9
Sun Jul 23 01:32:24 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 10.10.10.136 24:a1:60:2a:d0:e9 ESP-2AD0E9
Sun Jul 23 01:55:32 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 10.10.10.148 5c:e5:0c:d8:93:d2
Sun Jul 23 01:55:32 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 10.10.10.148 5c:e5:0c:d8:93:d2 lumi-acpartner-mcn02_miap93D2
Sun Jul 23 02:00:38 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 10.10.10.234 5c:e5:0c:49:73:e5
Sun Jul 23 02:00:38 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 10.10.10.234 5c:e5:0c:49:73:e5 lumi-acpartner-mcn02_miap73E5
Sun Jul 23 02:11:59 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 10.10.10.232 6a:06:4c:6b:47:4e
Sun Jul 23 02:11:59 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 10.10.10.232 6a:06:4c:6b:47:4e
Sun Jul 23 02:25:32 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 10.10.10.118 88:12:ac:fa:57:a7
Sun Jul 23 02:25:32 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 10.10.10.118 88:12:ac:fa:57:a7 xiaomidianshi
Sun Jul 23 02:27:51 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 10.10.10.190 88:46:04:46:81:02
Sun Jul 23 02:27:51 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 10.10.10.190 88:46:04:46:81:02 M2012K11AC
Sun Jul 23 02:42:00 2023 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 60:21:01:39:f6:4d
Sun Jul 23 02:42:00 2023 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 10.10.10.196 60:21:01:39:f6:4d
Sun Jul 23 02:42:00 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 10.10.10.196 60:21:01:39:f6:4d
Sun Jul 23 02:42:00 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 10.10.10.196 60:21:01:39:f6:4d OPPO-R11
Sun Jul 23 03:10:20 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 10.10.10.191 c6:2d:11:46:b9:aa
Sun Jul 23 03:10:20 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 10.10.10.191 c6:2d:11:46:b9:aa
Sun Jul 23 03:30:17 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 10.10.10.200 64:64:4a:a4:8e:68
Sun Jul 23 03:30:17 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 10.10.10.200 64:64:4a:a4:8e:68 xiaomi-woshi
Sun Jul 23 03:34:34 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 10.10.10.121 c0:ff:a8:93:41:3e
Sun Jul 23 03:34:34 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 10.10.10.121 c0:ff:a8:93:41:3e K662c
Sun Jul 23 03:35:15 2023 daemon.info pppd[11436]: No response to 5 echo-requests
Sun Jul 23 03:35:15 2023 daemon.notice pppd[11436]: Serial link appears to be disconnected.
Sun Jul 23 03:35:15 2023 daemon.info pppd[11436]: Connect time 2165.5 minutes.
Sun Jul 23 03:35:15 2023 daemon.info pppd[11436]: Sent 3489827000 bytes, received 1338736435 bytes.
Sun Jul 23 03:35:15 2023 daemon.info avahi-daemon[7245]: Withdrawing address record for 218.89.239.242 on pppoe-wan.
Sun Jul 23 03:35:15 2023 daemon.info avahi-daemon[7245]: Leaving mDNS multicast group on interface pppoe-wan.IPv4 with address 218.89.239.242.
Sun Jul 23 03:35:15 2023 daemon.info avahi-daemon[7245]: Interface pppoe-wan.IPv4 no longer relevant for mDNS.
Sun Jul 23 03:35:15 2023 daemon.notice ttyd[13364]: rops_handle_POLLIN_netlink: DELADDR
Sun Jul 23 03:35:15 2023 daemon.info avahi-daemon[7245]: Withdrawing address record for fe80::d0:4c50:c10:1550 on pppoe-wan.
Sun Jul 23 03:35:15 2023 daemon.notice ttyd[13364]: rops_handle_POLLIN_netlink: DELADDR
Sun Jul 23 03:35:15 2023 daemon.info avahi-daemon[7245]: Leaving mDNS multicast group on interface pppoe-wan.IPv6 with address fe80::d0:4c50:c10:1550.
Sun Jul 23 03:35:15 2023 daemon.notice netifd: Network device 'pppoe-wan' link is down
Sun Jul 23 03:35:15 2023 daemon.info avahi-daemon[7245]: Interface pppoe-wan.IPv6 no longer relevant for mDNS.
Sun Jul 23 03:35:15 2023 daemon.notice netifd: Interface 'wan' has lost the connection
Sun Jul 23 03:35:16 2023 daemon.err nmbd[12676]: [2023/07/23 03:35:16.375067,  0] ../../source3/nmbd/nmbd.c:60(terminate)
Sun Jul 23 03:35:16 2023 daemon.err nmbd[12676]:   Got SIGTERM: going down...
Sun Jul 23 03:35:16 2023 daemon.err nmbd[17519]: [2023/07/23 03:35:16.398614,  0] ../../lib/util/become_daemon.c:135(daemon_ready)
Sun Jul 23 03:35:16 2023 daemon.err nmbd[17519]:   daemon_ready: daemon 'nmbd' finished starting up and ready to serve connections
Sun Jul 23 03:35:16 2023 daemon.err smbd[17518]: [2023/07/23 03:35:16.410008,  0] ../../lib/util/become_daemon.c:135(daemon_ready)
Sun Jul 23 03:35:16 2023 daemon.err smbd[17518]:   daemon_ready: daemon 'smbd' finished starting up and ready to serve connections
Sun Jul 23 03:35:21 2023 daemon.notice pppd[11436]: Connection terminated.
Sun Jul 23 03:35:21 2023 daemon.info pppd[11436]: Connect time 2165.5 minutes.
Sun Jul 23 03:35:21 2023 daemon.info pppd[11436]: Sent 3489827000 bytes, received 1338736435 bytes.
Sun Jul 23 03:35:21 2023 daemon.debug ModemManager[17571]: hotplug: event reported: action=remove, name=pppoe-wan, subsystem=net
Sun Jul 23 03:35:21 2023 daemon.notice pppd[11436]: Modem hangup
Sun Jul 23 03:35:21 2023 daemon.info pppd[11436]: Exit.
Sun Jul 23 03:35:21 2023 daemon.notice netifd: Interface 'wan' is now down
Sun Jul 23 03:35:21 2023 daemon.notice netifd: Interface 'wan' is disabled
Sun Jul 23 03:35:21 2023 daemon.notice netifd: Interface 'wan6' is disabled
Sun Jul 23 03:35:21 2023 daemon.notice netifd: Interface 'wan' is enabled
Sun Jul 23 03:35:21 2023 daemon.notice netifd: Interface 'wan' is setting up now
Sun Jul 23 03:35:21 2023 daemon.notice netifd: Interface 'wan6' is enabled
Sun Jul 23 03:35:21 2023 daemon.notice netifd: Network device 'eth0' link is down
Sun Jul 23 03:35:21 2023 daemon.notice netifd: Interface 'wan' has link connectivity loss
Sun Jul 23 03:35:21 2023 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Sun Jul 23 03:35:21 2023 daemon.notice netifd: Interface 'wan' is now down
Sun Jul 23 03:35:21 2023 daemon.notice netifd: Interface 'wan' is disabled
Sun Jul 23 03:35:21 2023 daemon.notice netifd: Interface 'wan6' is disabled
Sun Jul 23 03:35:21 2023 daemon.notice netifd: Interface 'wan' is enabled
Sun Jul 23 03:35:21 2023 daemon.notice netifd: Interface 'wan6' is enabled
Sun Jul 23 03:35:24 2023 kern.info kernel: [129956.237550] igc 0000:01:00.0 eth0: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
Sun Jul 23 03:35:24 2023 daemon.notice netifd: Network device 'eth0' link is up
[... 85 lines cut ...]
Sun Jul 23 03:35:46 2023 daemon.err nlbwmon[19923]: Error relocating /usr/sbin/nlbwmon: avl_delete: symbol not found
Sun Jul 23 03:35:46 2023 daemon.err nlbwmon[19923]: Error relocating /usr/sbin/nlbwmon: avl_init: symbol not found
Sun Jul 23 03:35:46 2023 daemon.err nlbwmon[19923]: Error relocating /usr/sbin/nlbwmon: uloop_timeout_cancel: symbol not found
Sun Jul 23 03:35:46 2023 daemon.err nlbwmon[19923]: Error relocating /usr/sbin/nlbwmon: __calloc_a: symbol not found
Sun Jul 23 03:35:46 2023 daemon.err nlbwmon[19923]: Error relocating /usr/sbin/nlbwmon: avl_insert: symbol not found
Sun Jul 23 03:35:46 2023 daemon.err nlbwmon[19923]: Error relocating /usr/sbin/nlbwmon: uloop_run_timeout: symbol not found
Sun Jul 23 03:35:46 2023 daemon.err nlbwmon[19923]: Error relocating /usr/sbin/nlbwmon: uloop_fd_delete: symbol not found
Sun Jul 23 03:35:46 2023 daemon.err nlbwmon[19923]: Error relocating /usr/sbin/nlbwmon: usock: symbol not found
Sun Jul 23 03:35:46 2023 daemon.err nlbwmon[19923]: Error relocating /usr/sbin/nlbwmon: uloop_timeout_set: symbol not found
Sun Jul 23 03:35:46 2023 daemon.err nlbwmon[19923]: Error relocating /usr/sbin/nlbwmon: uloop_init: symbol not found
Sun Jul 23 03:35:46 2023 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan (pppoe-wan)
Sun Jul 23 03:36:01 2023 daemon.err nmbd[18919]: [2023/07/23 03:36:01.683255,  0] ../../source3/nmbd/nmbd_become_lmb.c:398(become_local_master_stage2)
Sun Jul 23 03:36:01 2023 daemon.err nmbd[18919]:   *****
Sun Jul 23 03:36:01 2023 daemon.err nmbd[18919]:
Sun Jul 23 03:36:01 2023 daemon.err nmbd[18919]:   Samba name server ISTOREOS is now a local master browser for workgroup WORKGROUP on subnet 10.10.10.252
Sun Jul 23 03:36:01 2023 daemon.err nmbd[18919]:
Sun Jul 23 03:36:01 2023 daemon.err nmbd[18919]:   *****
[... cut ...]
Sun Jul 23 03:37:26 2023 daemon.notice ddnsto[20664]: client init ok, username=45694fec-bf4f-4dba-ab55-da098f067c8e routerId=00d04c101550 version=3.0.2
Sun Jul 23 04:02:22 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 10.10.10.225 b6:af:a3:57:f9:79
Sun Jul 23 04:02:22 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 10.10.10.225 b6:af:a3:57:f9:79
Sun Jul 23 04:08:22 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 10.10.10.189 00:11:32:12:34:56
Sun Jul 23 04:08:22 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 10.10.10.189 00:11:32:12:34:56 LIU-NAS

jjm2473, 2023-07-24:
恢复出厂设置试试

shixin-679, 2023-07-28:
固件版本：iStoreOS 22.03.4 2023071416
平台：R5S
我也遇到了这个问题，我配置了微信推送。最近我每天收到大概10次以上的IP地址变化通知，导致了DDNS频繁更新，所以会出现公网访问老是得等TTL超时（之前提了一个Issues），看了下日志：
Fri Jul 28 17:38:08 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Fri Jul 28 17:38:08 2023 user.notice wrtbwmon: Restart for ifupdate of wan (pppoe-wan)
Fri Jul 28 17:38:08 2023 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Fri Jul 28 17:38:11 2023 daemon.err nlbwmon[16885]: Error loading shared library libubox.so.20210516: No such file or directory (needed by /usr/sbin/nlbwmon)
Fri Jul 28 17:38:11 2023 daemon.err nlbwmon[16885]: Error relocating /usr/sbin/nlbwmon: uloop_done: symbol not found
Fri Jul 28 17:38:11 2023 daemon.err nlbwmon[16885]: Error relocating /usr/sbin/nlbwmon: uloop_fd_add: symbol not found
Fri Jul 28 17:38:11 2023 daemon.err nlbwmon[16885]: Error relocating /usr/sbin/nlbwmon: avl_find: symbol not found
Fri Jul 28 17:38:11 2023 daemon.err nlbwmon[16885]: Error relocating /usr/sbin/nlbwmon: avl_delete: symbol not found
Fri Jul 28 17:38:11 2023 daemon.err nlbwmon[16885]: Error relocating /usr/sbin/nlbwmon: avl_init: symbol not found
Fri Jul 28 17:38:11 2023 daemon.err nlbwmon[16885]: Error relocating /usr/sbin/nlbwmon: uloop_timeout_cancel: symbol not found
Fri Jul 28 17:38:11 2023 daemon.err nlbwmon[16885]: Error relocating /usr/sbin/nlbwmon: __calloc_a: symbol not found
Fri Jul 28 17:38:11 2023 daemon.err nlbwmon[16885]: Error relocating /usr/sbin/nlbwmon: avl_insert: symbol not found
Fri Jul 28 17:38:11 2023 daemon.err nlbwmon[16885]: Error relocating /usr/sbin/nlbwmon: uloop_run_timeout: symbol not found
Fri Jul 28 17:38:11 2023 daemon.err nlbwmon[16885]: Error relocating /usr/sbin/nlbwmon: uloop_fd_delete: symbol not found
Fri Jul 28 17:38:11 2023 daemon.err nlbwmon[16885]: Error relocating /usr/sbin/nlbwmon: usock: symbol not found
Fri Jul 28 17:38:11 2023 daemon.err nlbwmon[16885]: Error relocating /usr/sbin/nlbwmon: uloop_timeout_set: symbol not found
Fri Jul 28 17:38:11 2023 daemon.err nlbwmon[16885]: Error relocating /usr/sbin/nlbwmon: uloop_init: symbol not found
Fri Jul 28 17:38:11 2023 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan_6 (pppoe-wan)
Fri Jul 28 17:38:11 2023 user.notice wrtbwmon: Restart for ifup of wan_6 (pppoe-wan)
Fri Jul 28 17:38:29 2023 daemon.info netdata[25695]: RRDSET: chart name 'net_drops.pppoe_wan' on host 'iStoreOS' already exists.
Fri Jul 28 17:38:39 2023 kern.info kernel: [451985.671978] rk_gmac-dwmac fe2a0000.ethernet eth0: Link is Down
Fri Jul 28 17:38:39 2023 daemon.notice netifd: Network device 'eth0' link is down
Fri Jul 28 17:38:39 2023 daemon.notice netifd: Interface 'wan' has link connectivity loss
Fri Jul 28 17:38:39 2023 daemon.info pppd[13510]: Terminating on signal 15
Fri Jul 28 17:38:39 2023 daemon.info pppd[13510]: Connect time 0.6 minutes.
Fri Jul 28 17:38:39 2023 daemon.info pppd[13510]: Sent 820739 bytes, received 25604162 bytes.
Fri Jul 28 17:38:39 2023 daemon.info avahi-daemon[7230]: Withdrawing address record for 100.64.74.192 on pppoe-wan.
Fri Jul 28 17:38:39 2023 daemon.info avahi-daemon[7230]: Leaving mDNS multicast group on interface pppoe-wan.IPv4 with address 100.64.74.192.
Fri Jul 28 17:38:39 2023 daemon.info avahi-daemon[7230]: Interface pppoe-wan.IPv4 no longer relevant for mDNS.
Fri Jul 28 17:38:39 2023 daemon.notice ttyd[14546]: rops_handle_POLLIN_netlink: DELADDR
Fri Jul 28 17:38:39 2023 daemon.notice ttyd[14546]: rops_handle_POLLIN_netlink: DELADDR
Fri Jul 28 17:38:39 2023 daemon.notice netifd: Network device 'pppoe-wan' link is down
Fri Jul 28 17:38:39 2023 daemon.notice netifd: Network alias 'pppoe-wan' link is down
Fri Jul 28 17:38:39 2023 daemon.notice netifd: Interface 'wan_6' has link connectivity loss
Fri Jul 28 17:38:39 2023 daemon.notice ttyd[14546]: rops_handle_POLLIN_netlink: DELADDR
Fri Jul 28 17:38:39 2023 daemon.info avahi-daemon[7230]: Interface pppoe-wan.IPv6 no longer relevant for mDNS.
Fri Jul 28 17:38:39 2023 daemon.info avahi-daemon[7230]: Leaving mDNS multicast group on interface pppoe-wan.IPv6 with address 240e:398:11b:9147:3eb3:8d0d:5542:38e8.
Fri Jul 28 17:38:39 2023 daemon.info avahi-daemon[7230]: Withdrawing address record for 240e:398:11b:9147:3eb3:8d0d:5542:38e8 on pppoe-wan.
Fri Jul 28 17:38:39 2023 daemon.err odhcp6c[14240]: Failed to send RS (Network unreachable)
Fri Jul 28 17:38:39 2023 daemon.err odhcp6c[14240]: Failed to send RELEASE message to ff02::1:2 (Network unreachable)
Fri Jul 28 17:38:39 2023 daemon.notice netifd: wan_6 (14240): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan_6" } (Permission denied)
Fri Jul 28 17:38:40 2023 daemon.err odhcp6c[14240]: Failed to send SOLICIT message to ff02::1:2 (Network unreachable)
Fri Jul 28 17:38:40 2023 daemon.notice netifd: Interface 'wan_6' is now down
Fri Jul 28 17:38:40 2023 daemon.info avahi-daemon[7230]: Withdrawing address record for 240e:39a:13c:71e0::1 on br-lan.
Fri Jul 28 17:38:40 2023 daemon.notice netifd: Interface 'wan_6' is disabled
Fri Jul 28 17:38:40 2023 daemon.warn dnsmasq[1]: no servers found in /tmp/resolv.conf.d/resolv.conf.auto, will retry
Fri Jul 28 17:38:40 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Fri Jul 28 17:38:40 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 10 names
Fri Jul 28 17:38:40 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
[... cut ...]
Fri Jul 28 17:56:55 2023 daemon.info netdata[25695]: RRDSET: chart name 'net_drops.pppoe_wan' on host 'iStoreOS' already exists.
Fri Jul 28 17:59:53 2023 daemon.notice ttyd[14546]: rops_handle_POLLIN_netlink: DELADDR
Fri Jul 28 18:00:44 2023 daemon.notice ttyd[14546]: rops_handle_POLLIN_netlink: DELADDR
Fri Jul 28 18:05:18 2023 user.info : luci: accepted login on / for root from 240e:39a:13c:7780:e195:8691:7e6b:13f9
Fri Jul 28 18:12:12 2023 daemon.notice ttyd[14546]: rops_handle_POLLIN_netlink: DELADDR
Fri Jul 28 18:13:03 2023 daemon.notice ttyd[14546]: rops_handle_POLLIN_netlink: DELADDR

shixin-679, 2023-07-28:
有几次发现IP地址变化时，WAN口指示灯好像是熄灭了再亮的
设备温度：
[image]

总是重连：
[image]

问了运营商，我的宽带最近在线都是几十个小时的，宽带没有异常离线

jjm2473, 2023-08-01:
@shixin-679 恢复出厂设置，然后不要随便升级系统的软件包，看你的日志很多报错是系统软件损坏。

jjm2473, 2023-08-01:
@shixin-679 `Fri Jul 28 17:56:18 2023 kern.info kernel: [453044.970211] rk_gmac-dwmac fe2a0000.ethernet eth0: Link is Down`，确定不是有人拔网线？或者光猫交换机重启了？

closed 2023-08-05 by jjm2473

shixin-679, 2023-08-05:
我的是光猫直连的软路由，软路由两个lan，一个接电脑，一个接无线路由器的，没有设备重启或拔网线过。。。
老大，恢复出厂设置那我的软路由配置能一键还原吗？好多插件配置起来太麻烦了

jjm2473, 2023-08-07:
后台不就有备份配置的功能吗