istoreos/istoreos #899: RTL8168网卡驱动错误问题

weizhituteng, 2023-07-23:
## 1. 详细叙述

### (1) 具体问题

A：2023071416版固件中，使用RTL8168网卡，系统会默认使用r8169的驱动，造成无法连接。我搜索了网上，也有很多Linux系统出现这个问题的，解决方法是替换网卡驱动。希望能在固件中尽快解决这个问题。

### (2) 路由器型号和固件版本

A：x86平台，固件版本iStoreOS 22.03.4，2023071416

### (3) 详细日志和/或截图

A：
[image]

[image]

jjm2473, 2023-07-24:
固件有集成r8168驱动，`lsmod |grep 8168` 看看。之所以保留8169驱动，是因为某些设备可能会用到。如果你要避免使用8169驱动，可以自己删掉或者卸载

closed 2023-07-24 by jjm2473

weizhituteng, 2023-07-24:
我通过命令查询，好像并没有集成r8168的专用驱动，8168是用的r8169的兼容驱动。麻烦您看下是不是这样的。
[image]

sheen197, 2024-09-27:
碰到了类似的问题。在wyse 5070上安装istoreos。版本iStoreOS 22.03.7 2024080210
因为wyse5070的网卡是8168，在istoreos中使用ethtool 发现 driver是r8169  firmware-version是 rtl8168h
结果的问题是网速上下行不对等明显。
台式机使用iperf或者homebox测速，下行（从软路由到台式机）1000M，上行（从台式机到软路由）60M

sheen197, 2024-09-27:
文件管理器查看lib/firmware，能看到rtl8168h相关的fw文件。在istoreos界面内的软件包界面的已安装列表，也能看到rtl8168相关组件。
但如果按照ethtool里的信息，那它就是反而先加载了r8169的驱动？
这时候是应该直接删除8169相关的驱动吗？