istoreos/istoreos #902: luci-theme-openwrt向导页面报错

binfoo, 2023-07-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
使用luci-theme-openwrt的主题，点击向导，页面报错

A：

### (2) 路由器型号和固件版本
<html>
<body>

目标平台 | rockchip/rk35xx
-- | --
固件版本 | iStoreOS 22.03.4 2023071416 / LuCI istoreos-22.03 branch git-23.103.23275-0193674
内核版本 | 5.10.176

</body>
</html>
A：

### (3) 详细日志和/或截图
[image]

A：

jjm2473, 2023-07-24:
这是官方主题的bug，原因是使用了旧html标准，不要用这个主题就行了。有问题就跟openwrt官方反馈

closed 2023-07-24 by jjm2473

jjm2473, 2023-07-24:
具体的代码位置

https://github.com/openwrt/luci/blob/f5efa3fbd09089d1131f867ab60b34d88e0ce122/themes/luci-theme-openwrt/luasrc/view/themes/openwrt.org/header.htm#L24

这一行限制了只能使用古老的html标准，不能使用html5。其他主题都不会限制只能使用老的html标准。istoreos部分页面使用vue3构建，必然会使用html5。

binfoo, 2023-07-24:
ok,把这个主题删了