istoreos/istoreos #903: 一个同时支持CX3，CX4及之后的Mellanox网卡的驱动，建议集成一下

kanepang, 2023-07-24:
[allegro0132](https://github.com/allegro0132)
/
[Openwrt-mlnx-ofed](https://github.com/allegro0132/Openwrt-mlnx-ofed)
a) 增加IB网络支持

通过安装本驱动，可以令你的OpenWRT路由器同时支持Infiniband和ETH网络，并能利用ipoib在ib网络上实现tcpip通信。本源码包正确编译运行后，输入
ip address
即可在输出中看到“ib0” ipoib网络适配器。
b) 在IB子网与ETH子网间通信
实例网络拓扑如下：外网通过eth接入wan端口，eth子网从lan端口接出，而ib子网从ib0端口输出。
需要先在IB子网的一台机器上开启opensm服务程序（暂不支持Openwrt），然后给ib0适配器配置ip地址，并将lan与ib0置于防火墙的同一区域下。这时ib适配器可与lan侧的eth设备通信，同理也能和wan侧通信。ib子网将与原有的eth网络通过openwrt三层功能连在一起，从而自由地访问eth网络。
c) 增强SRIOV兼容性
在虚拟机宿主开启SRIOV功能，并将VF直通进入OpenWRT虚拟机，能达到极佳的网络性能。但是OpenWRT集成的Mellanox系列网卡驱动存在问题，会导致OpenWRT系统在这种应用场景下无限重启，使用本源码包替代原有的mlx4，mlx5驱动，可解决这一问题。

jjm2473, 2023-07-28:
这么高端，是服务器网卡吗

kanepang, 2023-07-28:
算是最便宜的40G/G网卡吧，跟r86s带的10G网卡一样，只不过这个是台式电脑用的，pcie接口。黄鱼卖40元左右

Ta-152, 2023-08-19:
我用hyper-V建的istore虚拟机，用25G版MCX4121A建了个[外部]虚拟交换机给istore虚拟机用，网络状态显示“已降级 (SR-IOV 未运行)”，是不是也因为驱动问题呢？不过软路由里面倒是显示正常协商25G的

jjm2473, 2023-08-22:
SR-IOV应该是宿主机的功能，跟虚拟机无关

jjm2473, 2023-08-22:
看这个项目是backport给4.9内核用的，估计新内核本身就支持这些驱动

jjm2473, 2023-08-22:
Infiniband 是什么东西啊，我看op甚至都没有启用内核的这个功能

Ta-152, 2023-08-25:
> SR-IOV应该是宿主机的功能，跟虚拟机无关
我这个问题已经解决了，跟固件无关，属于hyper-v的毛病，开启了MAC地址欺骗就不能开SRIOV，就算成功启用了，两个PF间的通信也有问题，放弃用SRIOV了

mildnes, 2023-10-14:
@kanepang 你更新到最新版测试有没有集成了IB模式的sriov呢。。我刚踩了坑，debian12驱动不支持了

wenke313, 2025-05-13:
> > SR-IOV应该是宿主机的功能，跟虚拟机无关
> > 我这个问题已经解决了，跟固件无关，属于hyper-v的毛病，开启了MAC地址欺骗就不能开SRIOV，就算成功启用了，两个PF间的通信也有问题，放弃用SRIOV了
你也遇到两个PF通讯问题吗？无解的？