istoreos/istoreos #904: RTL8168网卡驱动错误问题

weizhituteng, 2023-07-24:
您好，我不太会用issues，在上一个issues关闭后我不太清楚如何再打开，所以又重新开了一个新的，请您见谅。

关联[https://github.com/istoreos/istoreos/issues/899](url)
## 1. 详细叙述

### (1) 具体问题

A：2023071416版固件中，使用RTL8168网卡，系统会默认使用r8169的驱动，造成无法连接。我搜索了网上，也有很多Linux系统出现这个问题的，解决方法是替换网卡驱动。希望能在固件中尽快解决这个问题。

### (2) 路由器型号和固件版本

A：x86平台，固件版本iStoreOS 22.03.4，2023071416

### (3) 详细日志和/或截图

A：
[image]

[image]

## 2. 后续
我按您在上一个issues里讲到的查询命令查找了一下，好像并没有集成r8168的专用驱动，8168是用的r8169的兼容驱动。麻烦您看下是不是这样的。

[image]

jjm2473, 2023-07-25:
直接在原来的issue回复就行了。

我找了下issue，之前是因为r8168驱动可能导致断网，而且r8168的优先级比8169高，所以才没集成 https://github.com/istoreos/istoreos/issues/320

jjm2473, 2023-07-25:
试试这个驱动，解压以后的ipk在istore那边可以安装，安装以后重启下。

[kmod-r8168_5.10.176+8.051.02-1_x86_64.ipk.zip](https://github.com/istoreos/istoreos/files/12157860/kmod-r8168_5.10.176%2B8.051.02-1_x86_64.ipk.zip)

weizhituteng, 2023-07-25:
您好，更换了您发的这个r8168驱动后，不报之前的"r8169 0000:03:00.0 eth4:rtl_rxtx_empty_cond == 0 (loop: 42, delay: 100)"的错误了。但是设备还是无法链接。日志里的信息是这样的。

[   31.743547] r8168: eth4: link up
[   31.743703] br-lan: port 4(eth4) entered blocking state
[   31.743707] br-lan: port 4(eth4) entered forwarding state
[  107.769301] nf_conntrack: default automatic helper assignment has been turned off for security reasons and CT-based  firewall rule not found. Use the iptables CT target to attach helpers instead.
[  132.140025] r8168: eth4: link up
[  132.140165] br-lan: port 4(eth4) entered blocking state
[  132.140171] br-lan: port 4(eth4) entered forwarding state

之后就是无限的循环以上三行，如下图：
[image]

这个问题就是更新了2023071416版固件后才出现的，在这更新之前用的2023060914版及之前的版本都无问题，正常链接的。我确定连接的设备是正常的（我把此设备直接接到AP上是可以连接到的，接在软路由就不行），连接的网线也尝试更换过。还麻烦再看下是什么造成了问题，谢谢。

jjm2473, 2023-07-26:
你确定这些网卡都加到wan或者lan了吗？

jjm2473, 2023-07-26:
执行`cat /etc/config/network`和`ip link`，把结果发出来。

weizhituteng, 2023-07-31:
上周后来有点忙，抱歉一直没有反馈。
我确定eth4加入到了LAN。
在昨天我发现有22.03.5-20230728版固件，升级后，eth4就能正常使用了。可能是22.03.4-2023071416固件有BUG造成的吧。目前0728版无问题，正常使用。
感谢您的一直帮助。

closed 2023-07-31 by jjm2473

789cn, 2023-12-27:
我J1900安装后怎么样都无法启动，一直卡R8169，无法启动也没法进SSH。

jjm2473, 2023-12-28:
@789cn  J1900可能不支持5.10内核，使用老的 21.02.3 版本固件，或者试试更新的 [23.05测试版固件](https://fw.koolcenter.com/iStoreOS/alpha/)

PlutoNameless, 2024-03-05:
> 试试这个驱动，解压以后的ipk在istore那边可以安装，安装以后重启下。
> 
> [kmod-r8168_5.10.176+8.051.02-1_x86_64.ipk.zip](https://github.com/istoreos/istoreos/files/12157860/kmod-r8168_5.10.176%2B8.051.02-1_x86_64.ipk.zip)

当前istoreos22.03.6已经是6.1版本了，是否还可以使用？

jjm2473, 2024-03-06:
@PlutoNameless 不再需要自己安装r8168驱动，在后台的“系统”-“杂项设置”-“驱动”里面选择r8168就行了，然后保存应用，重启系统。

> 当前istoreos22.03.6已经是6.1版本了，是否还可以使用？

哪来的6.1，你用的是官方固件吗？

PlutoNameless, 2024-03-06:
> @PlutoNameless 不再需要自己安装r8168驱动，在后台的“系统”-“杂项设置”-“驱动”里面选择r8168就行了，然后保存应用，重启系统。
> 
> > 当前istoreos22.03.6已经是6.1版本了，是否还可以使用？
> 
> 哪来的6.1，你用的是官方固件吗？
[image]
[image]
不好意思，我记错了内核版本；刚才看了下，确实是5.10；
我当前使用的版本是iStoreOS 22.03.6 2024030112
在尝试过你说的方法后，速度依旧没有得到改善；
此硬件我之前在另外一个Openwrt衍生系统上正常运行过，HomeLede linux内核5.4.138版本，不过在此衍生系统的其它版本，如5.10和6.1版本中，无法正常工作；
可能是8111需要特定的8168驱动版本才能正常工作？

PlutoNameless, 2024-03-06:
> @PlutoNameless 不再需要自己安装r8168驱动，在后台的“系统”-“杂项设置”-“驱动”里面选择r8168就行了，然后保存应用，重启系统。
> 
> > 当前istoreos22.03.6已经是6.1版本了，是否还可以使用？
> 
> 哪来的6.1，你用的是官方固件吗？

iStoreOS 22.03.6 2024030112
我这边又进行了一些测试，不管是使用8168还是8169，速度都处于异常；
[image]
[image]
[image]

jjm2473, 2024-03-07:
@PlutoNameless  在网络-接口-全局网络设置里面把“数据包引导”勾选上，然后重启下看看

uj6543fdjr6nb4, 2024-05-29:
“数据包引导”勾选上 没用的。我的试了。老样子