istoreos/istoreos #915: transmission远程无法访问WEB管理页面，已经做了端口转发，不知是不是设置问题，请教

ssupssup, 2023-07-28:
X64最新固件22.03.4 2023071416，想用transmission，发现远程无法访问WEB管理页面，已经做了端口转发，不知是不是设置问题，请教

18309871333, 2023-07-28:
今天刚升到22.03.5，安装完tr外网可以访问，发送一个任务马上无法连接了

scrymile, 2023-07-28:
> 今天刚升到22.03.5，安装完tr外网可以访问，发送一个任务马上无法连接了

是的，我也是，只要添加一个下载任务立马打不开。感觉开发者自己也不用transmission，估计也没更新版本的计划。还是尽量催一催吧

scrymile, 2023-07-29:
不是设置问题

18309871333, 2023-07-30:
用docker安装Tr4.0.3在挂上中文ui就能正常用了，暂时我就这么解决的

scrymile, 2023-08-03:
> 

怎么弄的

jjm2473, 2023-08-03:
我将一台x86升级到最新固件，然后重新安装transmission，然后使用首页的配置，分别选择两个不存在的文件夹，配置以后就自动启动了tr，然后添加磁力链接下载，没问题。
[image]

如果要使用已存在的文件夹，又不确定文件夹权限是否正确，那就按前面说的将“用户权限”和“组权限”都改成`root`。

scrymile, 2023-08-13:
> 我将一台x86升级到最新固件，然后重新安装transmission，然后使用首页的配置，分别选择两个不存在的文件夹，配置以后就自动启动了tr，然后添加磁力链接下载，没问题。 [image]
> 
> 如果要使用已存在的文件夹，又不确定文件夹权限是否正确，那就按前面说的将“用户权限”和“组权限”都改成`root`。

Mon Aug 14 02:14:58 2023 daemon.err transmission-daemon[5303]: [2023-08-13 18:14:58.862] Couldn't save file "/root/Configs/transmission/settings.json": Resource busy (variant.c:1226)

日志这么提示。

附上别人也碰到了这个问题https://github.com/openwrt/packages/issues/19376

jjm2473, 2023-08-15:
你链接的issue不是说了不影响运行吗？总之就是改成root

scrymile, 2023-08-17:
> 你链接的issue不是说了不影响运行吗？总之就是改成root

改成root，依然无法下载，下载的那个网页都打不开.....    pt站内也没显示我的Transmission有链接。。

scrymile, 2023-08-17:
web界面都打不开，怎么可能没问题哦

jjm2473, 2023-08-17:
那就看看系统日志，或者恢复出厂设置以后重新配置看能不能复现