istoreos/istoreos #922: n100 i226-V esxi 8.0中协商2.5G 但istoreos中显示1G

wuzhijun0301, 2023-07-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
n100 i226 esxi中显示协商2.5G 但是istoreos中显示1G

路由器是ax86u pro 接的2.5G口，esxi中已经显示是2.5g
[image]

istoreos中显示1g
[image]

Q：是否用关键词搜索了issue? (使用 "x" 选择)
没有类似的issue

## 2. 详细叙述

### (1) 具体问题
如上面描述

### (2) 路由器型号和固件版本
路由器：AX86U PRO 2.5g口
固件：22.03.5 2023072811
硬件：N100 i226-v 2.5g

### (3) 详细日志和/或截图

ethtool -i eth0
driver: e1000
version: 5.10.176
firmware-version: 
expansion-rom-version: 
bus-info: 0000:02:01.0
supports-statistics: yes
supports-test: yes
supports-eeprom-access: yes
supports-register-dump: yes
supports-priv-flags: no

root@iStoreOS:~ethtool eth0
Settings for eth0:
        Supported ports: [ TP ]
        Supported link modes:   10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: on
        MDI-X: off (auto)
        Supports Wake-on: d
        Wake-on: d
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: yes

搜索了所有的issue，实在没有思路了，望请帮忙看看

wuzhijun0301, 2023-07-30:
接触不是太久，不是非常熟悉，如果有建议或者尝试的解决方案，还请稍微说的详细一些，非常感谢！

jjm2473, 2023-08-03:
你确定中间没有经过交换机？

jjm2473, 2023-08-03:
esxi是直通网卡进去istoreos的吗，还是有虚拟网卡？AX86U是华硕的那个吗，不是硬路由吗？

jjm2473, 2023-08-03:
如果你想在虚拟机里面的网卡超过千兆，可以使用虚拟万兆网卡，也就是ESXI里面把虚拟网卡型号改成VMXNET3。

wuzhijun0301, 2023-08-03:
> 如果你想在虚拟机里面的网卡超过千兆，可以使用虚拟万兆网卡，也就是ESXI里面把虚拟网卡型号改成VMXNET3。

谢谢你说的这些点，我是华硕的硬路由AX86U Pro，软路由作为旁路由，在路由器里面，的确协商的是2.5G。

我发现我的网卡的确不是直通的，我现在使用eth0以及eth1口，eth0虚拟作为esxi的管理端口，eth1直通作为旁路由的口，这样在istoreos里面就显示2500M了。应该是我虚拟的网卡最多支持1G的缘故造成的

上面最后你提及了虚拟万兆网卡，我还没有尝试过，我可以试试你说的虚拟万兆，这样只要一个虚拟网卡作为管理+istoreos口

最后，感谢你的回复！

wuzhijun0301, 2023-08-03:
> 如果你想在虚拟机里面的网卡超过千兆，可以使用虚拟万兆网卡，也就是ESXI里面把虚拟网卡型号改成VMXNET3。

的确修改成VMXNET3就可以支持虚拟万兆了，之前是由于我将"操作系统版本"选择了“其他64位”导致只有E1000，而E1000只支持Gb。修改成3.x 4.x 5.x 6.x之类的64位，在创建虚拟网卡的时候，就会默认选择VMXNET3万兆了。

但是不知道为什么，我用homebox测试内网速度还是只有1G左右，按理说用mac无线测试至少可以到1000多才对

jjm2473, 2023-08-04:
自己用iperf3命令在每一个环节测速，找到瓶颈，问别人没用

rwang5986, 2024-04-29:
> > 如果你想在虚拟机里面的网卡超过千兆，可以使用虚拟万兆网卡，也就是ESXI里面把虚拟网卡型号改成VMXNET3。
> 
> 的确修改成VMXNET3就可以支持虚拟万兆了，之前是由于我将"操作系统版本"选择了“其他64位”导致只有E1000，而E1000只支持Gb。修改成3.x 4.x 5.x 6.x之类的64位，在创建虚拟网卡的时候，就会默认选择VMXNET3万兆了。
> 
> 但是不知道为什么，我用homebox测试内网速度还是只有1G左右，按理说用mac无线测试至少可以到1000多才对

同样问题，请问解决了么？

jjm2473, 2024-04-30:
这个题主的问题是因为使用了虚拟网卡，不是物理网卡直通，istoreos自然是用虚拟网卡的速度啊。