istoreos/istoreos #923: 建议完善photoprism功能，完善迁移说明文档

hbzhou530, 2023-07-31:
建议完善photoprism功能，完善迁移说明文档

1、功能小bug：
安装该应用，卸载重装后，新密码无效。
原因：旧配置文件都在，设置新密码后无法使用（已确认在docker容器的环境变量"PHOTOPRISM_ADMIN_PASSWORD"是新密码）。
建议完善photoprism的卸载脚本。

2、重装数据备份的问题
photoprism的索引数据 扫描时间很长。当重装ist、应用时，便涉及到备份还原问题。
如1的情况 重装时用的旧配置文件，但新旧秘密无效（不知道为什么浏览器能直接打开，可能有Cookie。换个新浏览器，密码无效）。
photoprism官方的还原文档没有纯docker的。

针对istoreos的用户环境（纯docker、无 MariaDB），急需photoprism索引的备份还原文档。

感谢大佬们开发istoreos这样简单易用的产品。

hbzhou530, 2023-08-03:
是的。你说的没错。PHOTOPRISM_ADMIN_PASSWORD只第一次部署时有效。
使用时，疏忽（没注意官方文档），部署完在这个位置又改一次密码，用了一段时间，发现只能本机登录。其他的登不上。

提示其他友人注意，密码在photoprism网页更改。不要忘记。

感觉在部署页能加个提示的话，更友好。

---- 回复的原邮件 ----
| 发件人 | Liangbin ***@***.***> |
| 日期 | 2023年08月03日 20:42 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [istoreos/istoreos] 建议完善photoprism功能，完善迁移说明文档 (Issue #923) |

PHOTOPRISM_ADMIN_PASSWORD 应该只在第一次启动才有用，如果后面自己改过密码了，那就用自己设置的密码。就算重新部署，只要旧配置文件还在，那就应该使用旧密码。这逻辑没啥问题吧？
用户的配置文件，一般在部署的时候就是可以选择或者自己填的，备份那个文件夹就行了。

jjm2473, 2023-08-04:
OK

jjm2473, 2023-08-04:
增加了提示了，卸载重装下，不会删除容器

closed 2023-08-04 by jjm2473