istoreos/istoreos #926: ARP路由表在线，但是实际上已经离线，ping不通

binfoo, 2023-08-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ *] 没有类似的issue

## 2. 详细叙述

路由表中设备还在线，首页也显示4个设备，但是实际上192.168.100.128是离线的

### (1) 具体问题

A：

### (2) 路由器型号和固件版本
[image]

A：

### (3) 详细日志和/或截图
[image]

[image]

A：

jjm2473, 2023-08-03:
ARP表有自己的刷新机制。设备下线难道还会主动通知路由器？肯定是有延迟的啊

closed 2023-08-03 by jjm2473

binfoo, 2023-08-04:
为啥延迟半天多？arp默认刷新不是30s吗？

jjm2473, 2023-08-04:
那你就抓包看看为什么arp记录还能维持