istoreos/istoreos #929: docker容器不能连接外网

wolf7689, 2023-08-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

已尝试恢复出厂设置，并且全新安装，问题依旧。具体表现为容器内可以ping通docker0联通的任何接口，包括网关。可以ping通宿主机的地址，但是与宿主机同一网段的任何机器都不能通。所以外网也不能通。

### (2) 路由器型号和固件版本

A：0728版本，装在pve中作为旁路由使用，通过网络向导配置

### (3) 详细日志和/或截图

A：traceroute跳到docker0网关就结束，或者也可以直接跳到宿主机内网地址

jjm2473, 2023-08-03:
有改过防火墙配置吗，防火墙有docker0的区域，默认允许转发到lan。还有，你的内网设备的默认网关是宿主机的ip吗？如果不是，那也ping不通的，需要NAT。

jjm2473, 2023-08-03:
这是一个网络路由的基本知识。你的docker0肯定跟内网不在一个网段，内网设备收到docker0过来的ping，那它要给谁回复呢？如果它自己没有路由规则，就会往它自己的默认网关回复，如果默认网关不是pve宿主机，那pve就收不到回复了，也就没法再回复给docker0。
有几个解决方案：
1. PVE配置NAT，对所有发到内网的数据都NAT
2. 在内网设备的网关设备配置静态路由，让docker0的ip段通过pve宿主机的内网ip路由
3. 在需要ping到的内网设备配置静态路由，路由规则同上一方案

closed 2023-08-03 by jjm2473

wolf7689, 2023-08-04:
感谢