istoreos/istoreos #931: 加载大量节点列表时，UI界面显示不全及加载过慢

ccwctv, 2023-08-04:
* [ ] 没有类似的issue

### (1) 具体问题

- 加载大量节点列表时，UI界面显示不全及加载过慢；
- 用passwall时发现，换用其他版本passwall或ssr-plus后，情景复现

### (2) 路由器型号和固件版本

- 型号：J4105
- 固件版本：iStoreOS 22.03.5 2023072811 / LuCI istoreos-22.03 branch git-23.103.23275-0193674

### (3) 详细日志和/或截图
[image]
[image]

jjm2473, 2023-08-05:
因为passwall还在使用服务端渲染，也就是旧版的luci，肯定会出现这个问题，不管你用什么固件。

closed 2023-08-05 by jjm2473

ccwctv, 2023-08-05:
> 因为passwall还在使用服务端渲染，也就是旧版的luci，肯定会出现这个问题，不管你用什么固件。

了解了，感谢解惑

jjm2473, 2023-08-07:
那你试试openwrt官方的固件，19以上的版本

ccwctv, 2023-08-07:
> 那你试试openwrt官方的固件，19以上的版本
现在换esir的最新固件，没有这个问题，OpenWrt R23.7.7，内核5.15.122