istoreos/istoreos #932: i210网卡会间歇性断网，而后重连

Colinelleo, 2023-08-05:
如题，有时会重连多次，有时重连后wan口会工作在百兆全双工，有时重连后是正常的千兆全双工。
日志如下。
Fri Aug  4 19:25:50 2023 daemon.info dnsmasq[1]: using only locally-known addresses for test
Fri Aug  4 19:25:50 2023 daemon.debug ModemManager[7283]: hotplug: event reported: action=add, name=tty38, subsystem=tty
Fri Aug  4 19:25:50 2023 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Fri Aug  4 19:25:50 2023 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Fri Aug  4 19:25:50 2023 daemon.info dnsmasq[1]: using only locally-known addresses for local
Fri Aug  4 19:25:50 2023 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Fri Aug  4 19:25:50 2023 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Fri Aug  4 19:25:50 2023 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Fri Aug  4 19:25:50 2023 daemon.warn dnsmasq[1]: no servers found in /tmp/resolv.conf.d/resolv.conf.auto, will retry
Fri Aug  4 19:25:50 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Fri Aug  4 19:25:50 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Fri Aug  4 19:25:50 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Fri Aug  4 19:25:50 2023 daemon.err ModemManager[7283]: hotplug: parent device sysfspath not found
Fri Aug  4 19:25:51 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Fri Aug  4 19:25:51 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Fri Aug  4 19:25:51 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Fri Aug  4 19:25:51 2023 daemon.warn dnsmasq-dhcp[1]: DHCP packet received on eth1 which has no address
Fri Aug  4 19:25:51 2023 kern.info kernel: [   20.921288] br-lan: port 1(eth1) entered blocking state
Fri Aug  4 19:25:51 2023 kern.info kernel: [   20.923142] br-lan: port 1(eth1) entered forwarding state
Fri Aug  4 19:25:51 2023 daemon.notice netifd: Network device 'eth1' link is up
Fri Aug  4 19:25:51 2023 daemon.err odhcp6c[11645]: Failed to send SOLICIT message to ff02::1:2 (Address not available)
Fri Aug  4 19:25:52 2023 daemon.notice netifd: Zerotier (10073): udhcpc: broadcasting discover
Fri Aug  4 19:25:52 2023 daemon.info avahi-daemon[7215]: Joining mDNS multicast group on interface eth0.IPv6 with address fe80::2f1:f3ff:fe1f:daa8.
Fri Aug  4 19:25:52 2023 daemon.info avahi-daemon[7215]: New relevant interface eth0.IPv6 for mDNS.
Fri Aug  4 19:25:52 2023 daemon.info avahi-daemon[7215]: Registering new address record for fe80::2f1:f3ff:fe1f:daa8 on eth0.*.
Fri Aug  4 19:25:52 2023 daemon.err dockerd[9666]: time="2023-08-04T11:25:52.350988666Z" level=warning msg="grpc: addrConn.createTransport failed to connect to {unix:///var/run/docker/containerd/containerd.sock  <nil> 0 <nil>}. Err :connection error: desc = \"transport: error while dialing: dial unix:///var/run/docker/containerd/containerd.sock: timeout\". Reconnecting..." module=grpc
Fri Aug  4 19:25:53 2023 daemon.err dockerd[9666]: time="2023-08-04T11:25:53Z" level=warning msg="containerd config version `1` has been deprecated and will be removed in containerd v2.0, please switch to version `2`, see https://github.com/containerd/containerd/blob/main/docs/PLUGINS.md#version-header"
Fri Aug  4 19:25:53 2023 daemon.err modprobe: failed to find a module named aufs
Fri Aug  4 19:25:53 2023 daemon.err dockerd[9666]: time="2023-08-04T11:25:53.519563171Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Fri Aug  4 19:25:53 2023 daemon.err dockerd[9666]: time="2023-08-04T11:25:53.521678022Z" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Fri Aug  4 19:25:53 2023 daemon.err dockerd[9666]: time="2023-08-04T11:25:53.526316556Z" level=error msg="failed to initialize a tracing processor \"otlp\"" error="no OpenTelemetry endpoint: skip plugin"
Fri Aug  4 19:25:53 2023 daemon.err dockerd[9666]: time="2023-08-04T11:25:53.581394576Z" level=warning msg="Your kernel does not support cgroup blkio weight"
Fri Aug  4 19:25:53 2023 daemon.err dockerd[9666]: time="2023-08-04T11:25:53.581435798Z" level=warning msg="Your kernel does not support cgroup blkio weight_device"
Fri Aug  4 19:25:53 2023 daemon.info modprobe: nf_conntrack is already loaded
Fri Aug  4 19:25:53 2023 daemon.info modprobe: nf_conntrack_netlink is already loaded
Fri Aug  4 19:25:53 2023 daemon.info avahi-daemon[7215]: Joining mDNS multicast group on interface docker0.IPv4 with address 172.17.0.1.
Fri Aug  4 19:25:53 2023 daemon.info avahi-daemon[7215]: New relevant interface docker0.IPv4 for mDNS.
Fri Aug  4 19:25:53 2023 daemon.info avahi-daemon[7215]: Registering new address record for 172.17.0.1 on docker0.IPv4.
Fri Aug  4 19:25:53 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 10.7.7.238 a0:36:9f:8d:0f:71
Fri Aug  4 19:25:53 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 10.7.7.238 a0:36:9f:8d:0f:71 Honey
Fri Aug  4 19:25:55 2023 daemon.notice netifd: Zerotier (10073): udhcpc: broadcasting discover
Fri Aug  4 19:25:56 2023 daemon.info procd: - init complete -
Fri Aug  4 19:25:58 2023 user.info : luci: accepted login on /admin/quickstart for root from 10.7.7.238
Fri Aug  4 19:25:59 2023 daemon.debug ModemManager[15931]: hotplug: event reported: action=add, name=ip6tnl0, subsystem=net
Fri Aug  4 19:25:59 2023 daemon.err ModemManager[15931]: hotplug: parent device sysfspath not found
Fri Aug  4 19:25:59 2023 daemon.debug ModemManager[15982]: hotplug: event reported: action=add, name=tunl0, subsystem=net
Fri Aug  4 19:25:59 2023 daemon.err ModemManager[15982]: hotplug: parent device sysfspath not found
Fri Aug  4 19:25:59 2023 daemon.debug ModemManager[16034]: hotplug: event reported: action=add, name=dummy0, subsystem=net
Fri Aug  4 19:25:59 2023 daemon.err ModemManager[16034]: hotplug: parent device sysfspath not found
Fri Aug  4 19:25:59 2023 daemon.debug ModemManager[16098]: hotplug: event reported: action=add, name=gre0, subsystem=net
Fri Aug  4 19:25:59 2023 daemon.err ModemManager[16098]: hotplug: parent device sysfspath not found
Fri Aug  4 19:25:59 2023 daemon.debug ModemManager[16152]: hotplug: event reported: action=add, name=gretap0, subsystem=net
Fri Aug  4 19:25:59 2023 daemon.err ModemManager[16152]: hotplug: parent device sysfspath not found
Fri Aug  4 19:25:59 2023 daemon.debug ModemManager[16208]: hotplug: event reported: action=add, name=erspan0, subsystem=net
Fri Aug  4 19:25:59 2023 daemon.err ModemManager[16208]: hotplug: parent device sysfspath not found
Fri Aug  4 19:25:59 2023 daemon.debug ModemManager[16268]: hotplug: event reported: action=add, name=ip6gre0, subsystem=net
Fri Aug  4 19:25:59 2023 daemon.err ModemManager[16268]: hotplug: parent device sysfspath not found
Fri Aug  4 19:25:59 2023 daemon.debug ModemManager[16320]: hotplug: event reported: action=add, name=siit0, subsystem=net
Fri Aug  4 19:25:59 2023 daemon.err ModemManager[16320]: hotplug: parent device sysfspath not found
Fri Aug  4 19:25:59 2023 daemon.debug ModemManager[16384]: hotplug: event reported: action=add, name=teql0, subsystem=net
Fri Aug  4 19:25:59 2023 daemon.err ModemManager[16384]: hotplug: parent device sysfspath not found
Fri Aug  4 19:25:59 2023 daemon.debug ModemManager[16460]: hotplug: event reported: action=add, name=br-lan, subsystem=net
Fri Aug  4 19:25:59 2023 daemon.err ModemManager[16460]: hotplug: parent device sysfspath not found
Fri Aug  4 19:26:00 2023 daemon.debug ModemManager[16519]: hotplug: event reported: action=add, name=ztks5z3jip, subsystem=net
Fri Aug  4 19:26:00 2023 daemon.err ModemManager[16519]: hotplug: parent device sysfspath not found
Fri Aug  4 19:26:00 2023 daemon.debug ModemManager[16603]: hotplug: event reported: action=add, name=docker0, subsystem=net
Fri Aug  4 19:26:00 2023 daemon.err ModemManager[16603]: hotplug: parent device sysfspath not found
Fri Aug  4 19:26:00 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
[... 371 lines cut ...]
Sat Aug  5 04:07:36 2023 daemon.err nmbd[21472]: [2023/08/05 04:07:36.817008,  0] ../../source3/nmbd/nmbd.c:60(terminate)
Sat Aug  5 04:07:36 2023 daemon.err nmbd[21472]:   Got SIGTERM: going down...
Sat Aug  5 04:07:36 2023 daemon.err nmbd[1894]: [2023/08/05 04:07:36.849420,  0] ../../lib/util/become_daemon.c:135(daemon_ready)
Sat Aug  5 04:07:36 2023 daemon.err nmbd[1894]:   daemon_ready: daemon 'nmbd' finished starting up and ready to serve connections
[... cut ...]
Sat Aug  5 04:07:43 2023 daemon.info pppd[1629]: Remote message: Authentication success,Welcome!
Sat Aug  5 04:07:43 2023 daemon.notice pppd[1629]: PAP authentication succeeded
Sat Aug  5 04:07:43 2023 daemon.notice pppd[1629]: peer from calling number 6C:14:6E:26:C4:AD authorized
Sat Aug  5 04:07:43 2023 daemon.notice ttyd[11293]: rops_handle_POLLIN_netlink: DELADDR
Sat Aug  5 04:07:43 2023 daemon.info avahi-daemon[7215]: Joining mDNS multicast group on interface pppoe-wan.IPv4 with address 121.57.226.218.
主要是有 NIC link is down. 
请问大佬是驱动问题吗？又或是硬件问题，有可能解决吗？

jjm2473, 2023-08-07:
看看是不是过热，或者换网线试试

jjm2473, 2023-08-18:
什么情况，解决了吗

closed 2023-10-25 by jjm2473