istoreos/istoreos #934: docker桥接网络模式无法访问互联网

makerdegit, 2023-08-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：docker容器的网络用桥接模式就断网了，局域网还可以连接到容器的web界面，最容易复现的就是istore里安装迅雷，选择非host模式，创建的镜像就是桥接模式的，这时候打开迅雷就无法登录。

### (2) 路由器型号和固件版本

A：x86物理机 
iStoreOS 22.03.5 2023080410
不止这个版本有这个问题

### (3) 详细日志和/或截图

A：
防火墙没改过设置
[image]
docker网口原本没有，我打开我以前一个虚拟机里的系统看见有这个网口，我就自己创建了，但有没有这个网口貌似没什么区别，我忘记没有的情况是不是会导致docker创建容器时用到端口映射就直接不能创建成功，是遇到过这个情况
[image]

makerdegit, 2023-08-06:
问题很好复现，麻烦大佬解决一下

jjm2473, 2023-08-07:
你这个是因为旁路由的情况下，流量不是从wan口出去，而是从lan口出去，lan口默认没开NAT。

所以解决方法也很简单，在防火墙页面，把lan口的“IP动态伪装”勾选上

closed 2023-08-07 by jjm2473