istoreos/istoreos #941: 求宝哥团队添加J6413 核显驱动!!谢谢

yaoyao19890513, 2023-08-09:
root@iStoreOS:~# lspci -tvnn
-[0000:00]-+-00.0  Intel Corporation Device [8086:4522]
           +-02.0  Intel Corporation Device [8086:4555]
           +-08.0  Intel Corporation Device [8086:4511]
           +-14.0  Intel Corporation Device [8086:4b7d]
           +-14.2  Intel Corporation Device [8086:4b7f]
           +-16.0  Intel Corporation Device [8086:4b70]
           +-17.0  Intel Corporation Device [8086:4b63]
           +-1c.0-[01]----00.0  Intel Corporation Device [8086:125c]
           +-1c.1-[02]----00.0  Intel Corporation Device [8086:125c]
           +-1c.2-[03]----00.0  Intel Corporation Device [8086:125c]
           +-1c.3-[04]----00.0  Intel Corporation Device [8086:125c]
           +-1c.4-[05]----00.0  Intel Corporation Device [8086:125c]
           +-1c.6-[06]----00.0  Intel Corporation Device [8086:125c]
           +-1f.0  Intel Corporation Device [8086:4b00]
           +-1f.3  Intel Corporation Device [8086:4b58]
           +-1f.4  Intel Corporation Device [8086:4b23]
           \-1f.5  Intel Corporation Device [8086:4b24]

jjm2473, 2023-08-11:
参考这个 https://github.com/istoreos/istoreos/issues/942

closed 2023-08-11 by jjm2473

jjm2473, 2023-08-11:
哦，已经支持了，只是硬解的配置不一样 https://github.com/istoreos/istoreos/issues/716#issuecomment-1599049259