istoreos/istoreos #944: 同一文件管理挂载的webdav无法重命名文件夹或者在其中另存文件

CG1995, 2023-08-10:
Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x ] 没有类似的issue

2. 详细叙述

(1) 具体问题

A：同一文件管理挂载硬盘后，无法在这个挂载盘里重命名文件夹，比如在windows上显示：操作无法完成，因为其中的文件夹或文件已在另一程序中打开，我很确定我没有在别的地方打开这个文件夹，文件夹是新建的还热乎

 (2) 路由器型号和固件版本

A：r5s和r5c

 (3) 详细日志和/或截图

A：
[image]

jjm2473, 2023-08-11:
建议使用samba，webdav不适合写

CG1995, 2023-08-12:
> 建议使用samba，webdav不适合写

samba无法外网挂载，我看我的nas还有alist挂载的webdav都能正常读写，是不是我哪里设置的不对造成的

jjm2473, 2023-08-13:
samba其实是可以外网挂载的，只是可能要端口转发，还有挂载的时候需要指定端口和共享文件夹名称，而不是只填服务器地址

CG1995, 2023-08-14:
> samba其实是可以外网挂载的，只是可能要端口转发，还有挂载的时候需要指定端口和共享文件夹名称，而不是只填服务器地址

原来如此，十分感谢哈！

1klakla1, 2023-08-31:
> samba其实是可以外网挂载的，只是可能要端口转发，还有挂载的时候需要指定端口和共享文件夹名称，而不是只填服务器地址

大佬，0825版本的固件 samba 有BUG，播放视频几分钟就卡掉挂在，必须重启路由才能解决。然后无限循环。

jjm2473, 2023-08-31:
@1klakla1 上一版本没问题吗？

1klakla1, 2023-08-31:
iStoreOS 21.02.3 2023042111
我上一个版本用的是这个版本号，没有问题。
更新了固件就出问题了，然后我就回滚快照了。测试了好几遍确定是新固件导致的。

jjm2473, 2023-08-31:
@1klakla1 不保留配置升级试过吗