istoreos/istoreos #948: 迅雷下载无法配置多个下载路径

xmasrain, 2023-08-11:
1.问题：istoreos中的docker应用迅雷下载，在配置“下载路径”的时候只能填写一个路径，这对于有多个盘符的用户则无法使用，若强行把路径设置为“/mnt”，则到应用中会仅识别为4M大小的空间，无法下载任何文件。

Q：是否用关键词搜索了issue? (使用 "x" 选择)
A：没有类似的issue

2. 详细叙述

istoreos中的docker应用迅雷，在配置“下载路径”的时候只能填写一个路径，这对于有多个盘符的用户则无法使用，若强行把路径设置为“/mnt”，则到应用中会仅识别为4M大小的空间，无法下载任何文件。

(2) 路由器型号和固件版本

A：设备型号
GEIT UT3000F
固件版本
iStoreOS 22.03.5 2023080410
内核版本
5.10.176

 (3) 详细日志和/或截图
[image]
[image]

jjm2473, 2023-08-22:
说明你的/mnt/nas8t根本没挂载啊，要么就是迅雷启动时还没挂载，可以在系统-杂项设置里面配置等待硬盘

closed 2023-08-22 by jjm2473

llonline, 2023-11-10:
我也碰到同样问题。硬盘已经挂载了，但是就是无法使用。之前一直好的，但是怎么操作的已经忘记了，换了一块大硬盘，就不行了。

jjm2473, 2023-11-10:
杂项设置里面配置等待硬盘

lff110, 2024-07-03:
不行就重启一下设备也可以