istoreos/istoreos #949: 防火墙修改区域设置后自动回跳修改前的设置

chenxv399, 2023-08-11:
## 1.关于你要提交的问题

在修改防火墙的连接规则并保存设置后（如将wan口入站数据由默认的丢弃改为接收），此时任意修改接口wan的设置 或者是重新连接wan口 再或者是重启设备后，防火墙修改的设置自动跳回默认设置（如此时的wan口入站数据自动重新变回丢弃）。

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在修改防火墙的连接规则并保存设置后（如将wan口入站数据由默认的丢弃改为接收），此时任意修改接口wan的设置 或者是重新连接wan口 再或者是重启设备后，防火墙修改的设置自动跳回默认设置（如此时的wan口入站数据自动重新变回丢弃）。

### (2) 路由器型号和固件版本

A：FastRhino R66s  
iStoreOS 22.03.5 2023080410

### (3) 详细日志和/或截图

A：
Fri Aug 11 17:04:09 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Fri Aug 11 17:04:09 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Fri Aug 11 17:04:09 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 4 names
Fri Aug 11 17:04:09 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
**Fri Aug 11 17:05:24 2023 user.notice wan_drop: current proto pppoe
Fri Aug 11 17:05:24 2023 user.notice wan_drop: set firewall drop input from wan**
Fri Aug 11 17:05:25 2023 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Fri Aug 11 17:05:25 2023 daemon.info dnsmasq[1]: using nameserver 119.29.29.29 53
Fri Aug 11 17:05:25 2023 daemon.info dnsmasq[1]: using nameserver 223.5.5.5 53
Fri Aug 11 17:05:25 2023 daemon.info dnsmasq[1]: using nameserver 2400:3200:baba::1 53
Fri Aug 11 17:05:25 2023 daemon.info dnsmasq[1]: using nameserver 2402:4e00:: 53
Fri Aug 11 17:05:25 2023 daemon.info dnsmasq[1]: using only locally-known addresses for test
Fri Aug 11 17:05:25 2023 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Fri Aug 11 17:05:25 2023 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Fri Aug 11 17:05:25 2023 daemon.info dnsmasq[1]: using only locally-known addresses for local
Fri Aug 11 17:05:25 2023 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Fri Aug 11 17:05:25 2023 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Fri Aug 11 17:05:25 2023 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Fri Aug 11 17:05:26 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Fri Aug 11 17:05:26 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Fri Aug 11 17:05:26 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 4 names
Fri Aug 11 17:05:26 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses

[image]
[image]

jjm2473, 2023-08-18:
早就有人发过了啊  https://github.com/istoreos/istoreos/issues/632

为了防止外网攻击，pppoe拨号时自动禁止wan口入站。

1. 可以自己在流量规则配置入站允许，
3. 或者在istore安装systool，里面可以关掉自动wan_drop脚本。

closed 2023-08-18 by jjm2473

gs314520, 2023-09-11:
打开etc/config目录下的uhttpd文件，找到option rfc1918_filter '1'，把1改成0即可

BL1ACH, 2024-04-02:
我也遇到这个情况了，真是欲哭无泪啊。外地工作没法控制家里设备，打视频教老婆我都快急死了。